coolsnowwolf/lede #5231: 编译报错  看不懂代码 麻烦帮忙看下

IvanSolis1989, 2020-08-01:
树莓派4编译固件报错 
if [ -d /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo ]; then (cd /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo; find ./ > /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo.files);        SHELL= flock /home/ivan/lede/tmp/.staging-dir.flock -c ' mv /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo.files /home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/packages/UnblockNeteaseMusicGo.list && cp -fpR /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo/* /home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/; '; fi
rm -rf /home/ivan/lede/tmp/stage-UnblockNeteaseMusicGo
touch /home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.UnblockNeteaseMusicGo_installed
make[3]: Leaving directory '/home/ivan/lede/package/lean/UnblockNeteaseMusicGo'
time: package/lean/UnblockNeteaseMusicGo/compile#30.54#19.50#19.40
make[2]: Leaving directory '/home/ivan/lede'
package/Makefile:107: recipe for target '/home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ivan/lede'
Build failed - please re-run with -j1 to see the real error message
/home/ivan/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

IvanSolis1989, 2020-08-01:
这是单线程编译的结果
install -d -m0755 /home/ivan/lede/bin/packages/aarch64_cortex-a72/packages
/home/ivan/lede/scripts/ipkg-build -c -o 0 -g 0 /home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/smartmontools-7.1/ipkg-aarch64_cortex-a72/smartmontools /home/ivan/lede/bin/packages/aarch64_cortex-a72/packages
Packaged contents of /home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/smartmontools-7.1/ipkg-aarch64_cortex-a72/smartmontools into /home/ivan/lede/bin/packages/aarch64_cortex-a72/packages/smartmontools_7.1-3_aarch64_cortex-a72.ipk
echo "smartmontools" >> /home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/smartmontools.default.install
make[3]: Leaving directory '/home/ivan/lede/feeds/packages/utils/smartmontools'
time: package/feeds/packages/smartmontools/compile#11.73#9.82#20.84
make[3]: Entering directory '/home/ivan/lede/package/firmware/cypress-firmware'
touch /home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/cypress-firmware-v5.4.18-2020_0402/.prepared_608b9199b723de1d1db962cde3e0bda7_6664517399ebbbc92a37c5bb081b5c53_check
. /home/ivan/lede/include/shell.sh; unzip -q -p /home/ivan/lede/dl/cypress-fmac-v5.4.18-2020_0402.zip cypress-firmware-v5.4.18-2020_0402.tar.gz | gzip -dc | tar -C /home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/cypress-firmware-v5.4.18-2020_0402 -xf -
error:  cannot open zipfile [ /home/ivan/lede/dl/cypress-fmac-v5.4.18-2020_0402.zip ]
        Permission denied

gzip: stdin: unexpected end of file
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:51: recipe for target '/home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/cypress-firmware-v5.4.18-2020_0402/.prepared_608b9199b723de1d1db962cde3e0bda7_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/ivan/lede/build_dir/target-aarch64_cortex-a72_musl/cypress-firmware-v5.4.18-2020_0402/.prepared_608b9199b723de1d1db962cde3e0bda7_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/ivan/lede/package/firmware/cypress-firmware'
time: package/firmware/cypress-firmware/compile#0.04#0.32#0.37
package/Makefile:111: recipe for target 'package/firmware/cypress-firmware/compile' failed
make[2]: *** [package/firmware/cypress-firmware/compile] Error 2
make[2]: Leaving directory '/home/ivan/lede'
package/Makefile:107: recipe for target '/home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/ivan/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ivan/lede'
/home/ivan/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

hyird, 2020-08-02:
rm -rf dl
make clean

Cathgao, 2020-08-02:
```
error: cannot open zipfile [ /home/ivan/lede/dl/cypress-fmac-v5.4.18-2020_0402.zip ]
Permission denied
```
你该不会用的root编译吧

Albelt, 2021-03-16:
树莓派4B，遇到了一样的错误，普通用户编译（没有用sudo），请问你解决这个问题了吗？

closed 2021-06-20 by coolsnowwolf