coolsnowwolf/lede #5238: 编译的时候提示缺少libjpeg-turbo，不知道具体应该安装哪个包

ricksuzade-maker, 2020-08-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
编译准备的时候提示：

WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libjpeg-turbo', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libjpeg-turbo', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a build dependency on 'libjpeg-turbo', which does not exist

### (1) 具体问题

A：提示缺少libjpeg-turbo

### (2) 路由器型号和固件版本

A：x64

qqsir-dev, 2020-08-02:
我也遇到了这个问题，不知道如何安装libjpeg-turbo

Cathgao, 2020-08-02:
这个也不是报错啊，不能编译吗？

dongjuanyong, 2020-08-03:
https://github.com/openwrt/packages/commit/8abf547919846941f429907b2f61aec6055760d0
https://github.com/openwrt/packages/commit/73d6ad6e1a984b18fb33cb25f24f5081cf4b5492
OpenWrt官方的packages feed更换了libjpeg的包
https://github.com/openwrt/telephony/commit/0e055c4657eb454959a45215775298b5114ef286
于是telephony feed的freeswitch也换了dep
lean大用的官方telephony feed和自己的packages feed，所以出现了dep err，你要是不用freeswitch的话可以无视

diabolic00, 2020-08-03:
> [openwrt/packages@8abf547](https://github.com/openwrt/packages/commit/8abf547919846941f429907b2f61aec6055760d0)
> [openwrt/packages@73d6ad6](https://github.com/openwrt/packages/commit/73d6ad6e1a984b18fb33cb25f24f5081cf4b5492)
> OpenWrt官方的packages feed更换了libjpeg的包
> [openwrt/telephony@0e055c4](https://github.com/openwrt/telephony/commit/0e055c4657eb454959a45215775298b5114ef286)
> 于是telephony feed的freeswitch也换了dep
> lean大用的官方telephony feed和自己的packages feed，所以出现了dep err，你要是不用freeswitch的话可以无视

大哥~怎么无视？小白全靠L大的教程才走到这步，求教怎么改才能无视这条，./scripts/feeds install -a，这条以后就全是err了。小白顿首

minlang112, 2020-08-04:
> 大哥~怎么无视？小白全靠L大的教程才走到这步，求教怎么改才能无视这条，./scripts/feeds install -a，这条以后就全是err了。小白顿首

如果不使用 freeswitch 这个组件，可以无视该报错

mysteryman2045, 2020-08-05:
> 大哥~怎么无视？小白全靠L大的教程才走到这步，求教怎么改才能无视这条，./scripts/feeds install -a，这条以后就全是err了。小白顿首

昨天编译的时候也发现这个WARNING，但是不至于./scripts/feeds install -a后全是err。无视，继续编译就OK了。如果是首次编译到这里出现err，都还没到make menuconfig，显然是前面的步骤出了问题，可能有些该获取、安装的资源没有正常获取安装到，继续重复前边的步骤就好。如果是再次编译，make clean一下，按照L大的二次编译流程再走一遍。

diabolic00, 2020-08-05:
> 昨天编译的时候也发现这个WARNING，但是不至于./scripts/feeds install -a后全是err。无视，继续编译就OK了。如果是首次编译到这里出现err，都还没到make menuconfig，显然是前面的步骤出了问题，可能有些该获取、安装的资源没有正常获取安装到，继续重复前边的步骤就好。如果是再次编译，make clean一下，按照L大的二次编译流程再走一遍。

明白了谢谢大佬

closed 2020-08-10 by coolsnowwolf