coolsnowwolf/lede #5247: SSR+的广告数据库无法更新；实时流量监测里的网速不准确

willaytoun, 2020-08-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：SSR+的广告数据库无法更新；实时流量监测里的网速不准确，实际有几MB的速度但是显示只有几KB或者几B

### (2) 路由器型号和固件版本

A：
型号 | Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4 Core 4 Thread (CpuMark : 33947.343042 Scores)
-- | --
CPU 信息 | 1606.322 MHz +37.0°C (high = +105.0°C, crit = +105.0°C)
固件版本 | OpenWrt R20.7.20 built by Lean & KFERM / LuCI Master (git-20.191.36863-eee6bae)
内核版本 | 5.4.51

dolphinpaopao, 2020-08-03:
这里已经删除ssr，请到新的项目下反馈相关问题。另外今天ssr已经更新版本解决了反广告订阅地址问题。敬请自行编译更新。

willaytoun, 2020-08-03:
请问新项目在哪啊？

yangabc123, 2020-08-03:
ssr的去广告地址已经失效，可以自己替换

LoogCP, 2020-08-04:
adbyby的规则（内置和自己添加的）也都无法更新

willaytoun, 2020-08-04:
流量监测大家那边是准确吗？

LoogCP, 2020-08-05:
> 流量监测大家那边是准确吗？

没有在意过流量监测，

hcym, 2020-08-05:
除了ssr，其它根本不编译

willaytoun, 2020-08-05:
满速下片显示只有几KB，我想应该是不准吧(✿◡‿◡)

willaytoun, 2020-08-05:
楼上说的SSR+新项目地址大家可以发一下吗

Cathgao, 2020-08-05:
如果使用了硬件加速，流量统计就会不准确

closed 2021-06-20 by coolsnowwolf