coolsnowwolf/lede #5248: r2s

r1172464137, 2020-08-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r2s固件不大稳定，近期编译出来的固件又经常获取不到ip

### (2) 路由器型号和固件版本

A：r2s

### (3) 详细日志

A：无日志

mjyhj, 2020-08-03:
源码里没有R2S啊，杂交的只能自己解决

EnnawYang, 2020-08-03:
> 源码里没有R2S啊，杂交的只能自己解决

有

r1172464137, 2020-08-03:
有没有人遇到过和我差不多的问题

1715173329, 2020-08-03:
请首先检查您的储存介质。
此外，根据部分人所述，需要在开机之前将网线接入 LAN 口，方可正常获取 DHCP 下发的 IP，您可以试试看。

minlang112, 2020-08-04:
[R2S:更换接口物理设置 #5028](https://github.com/coolsnowwolf/lede/pull/5028)

代码库里 R2S的LAN WAN已经调整为和硬件描述一致，和以前的杂交野生固件未必一样，自己注意一下

r1172464137, 2020-08-04:
> 请首先检查您的储存介质。
> 此外，根据部分人所述，需要在开机之前将网线接入 LAN 口，方可正常获取 DHCP 下发的 IP，您可以试试看。

不是卡的问题，昨天上京东买了两张不同牌子的卡，今天到了，下午重新试了一下，但同样是刷进去获取不到ip，刷了老固件就可以了。

r1172464137, 2020-08-04:
> [R2S:更换接口物理设置 #5028](https://github.com/coolsnowwolf/lede/pull/5028)
> 
> 代码库里 R2S的LAN WAN已经调整为和硬件描述一致，和以前的杂交野生固件未必一样，自己注意一下

我知道网口互换了，但还是无法获取。今天全新环境编译了一次依然无法获取到ip

r1172464137, 2020-08-04:
可能是我的硬件体制问题吧，我去换个货试试

minlang112, 2020-08-04:
> 可能是我的硬件体制问题吧，我去换个货试试

有TTL线和转接板的话，可以试试TTL接上去，看启动时候内核打印的日志有没有异常

closed 2020-08-07 by r1172464137