coolsnowwolf/lede #5251: 启用IPV6的情况下，外网会自动断网

keivenliao, 2020-08-04:
和大家交流一下，看看是否有和我同样的问题
当前状态：
1、x86版本，最新代码编译，只编译了 如 ddns、SSP、等少量应用
2、启用了IPV6，同时启用Adguardhome做的广告过滤

症状：
重启后能正常使用1、2天，然后会突然断外网，由于我在公司，就让家人重启解决问题，没有保留到日志
我在家遇到过一次，貌似是外网pppoe成功，有ipv4地址，没有ipv6 PD信息，导致无法上网（老婆大人发飙，没能保存下日志）

不知大家是否有类似情况？

hcym, 2020-08-04:
没用adg，用的ipv6，基本正常，不过路由和光猫都设置成一天一启，不然运营商换IP后ipv6会失效

keivenliao, 2020-08-04:
> 没用adg，用的ipv6，基本正常，不过路由和光猫都设置成一天一启，不然运营商换IP后ipv6会失效

我也怀疑是运营商换IP造成的，但是如果换IP就断网，应该也是有点问题的。
今晚回去看看日志啥问题。

Cathgao, 2020-08-04:
我的ipv6时不时会被运营商换掉，接口里的wan6显示有两个ipv6地址和一个旧的ipv6-pd造成ipv6链路断掉，要手动点连接。但ipv4不受影响，不会断网

minlang112, 2020-08-04:
该原因是由于运营商定时更换PD导致的，建议摸清规律后在凌晨定时重启。

xiehuayi75, 2020-08-04:
mwan3和ipv6都有冲突嫌疑，我也是每天一换ip就不拨号了，要手动链接wan口，真蛋疼

fhbyljj, 2020-09-12:
0 5 * * * /sbin/ifup wan
0 5 * * * /sbin/ifup wan6

LoogCP, 2020-09-14:
我也有这个问题，开了ipv6，过段时间（不确定多长），就会断网，表现和WAN口断开连接是一样的，也无法ping出去，手动点击WAN接口的连接后正常，我先试试上面那个重启wan和WAN6那个定时任务

fhbyljj, 2021-04-24:
我也发生了这个问题

LoogCP, 2021-04-24:
我用cron写了自动重启网络就好了/sbin/ifup wan/sbin/ifup wan6根据需要在crontab里面写入，我是用的8小时重启一次-------- 原始邮件 --------发件人： fhbyljj ***@***.***>日期： 2021年4月24日周六 17:04收件人： coolsnowwolf/lede ***@***.***>抄送： LoogCP ***@***.***>, Comment ***@***.***>主    题： Re: [coolsnowwolf/lede] 启用IPV6的情况下，外网会自动断网 (#5251)
我也发生了这个问题

closed 2021-06-20 by coolsnowwolf