coolsnowwolf/lede #5254: 编译报错

zhangzenglin, 2020-08-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [✓] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用Ubuntu 20.04 环境按照步骤编译报错

### (2) 路由器型号和固件版本

A：x86，64位

### (3) 详细日志

A：g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ilto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/home/zhangzenglin/lede/staging_di
r/host/include -I/home/zhangzenglin/lede/staging_dir/host/include -I/home/zhangzenglin/lede/staging_dir/host/include  -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o lto/lto-object.o -MT lto/lto-object.o -MMD -MP -MF lto/.deps/lto-object.TPo /home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto/lto-object.c
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ilto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/home/zhangzenglin/lede/staging_di

r/host/include -I/home/zhangzenglin/lede/staging_dir/host/include -I/home/zhangzenglin/lede/staging_dir/host/include  -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o lto/lto-partition.o -MT lto/lto-partition.o -MMD -MP -MF lto/.deps/lto-partition.TPo /home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto/lto-partition.c
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ilto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/home/zhangzenglin/lede/staging_dir/host/include -I/home/zhangzenglin/lede/staging_dir/host/include -I/home/zhangzenglin/lede/staging_dir/host/include  -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o lto/lto-symtab.o -MT lto/lto-symtab.o -MMD -MP -MF lto/.deps/lto-symtab.TPo /home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto/lto-symtab.c
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o lto1 \
        lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -L/home/zhangzenglin/lede/staging_dir/host/lib -L/home/zhangzenglin/lede/staging_dir/host/lib -L/home/zhangzenglin/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
collect2: error: ld returned 1 exit status
make[5]: *** [/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/lto/Make-lang.in:81: lto1] Error 1
make[5]: Leaving directory '/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc'
make[4]: *** [Makefile:4250: all-gcc] Error 2
make[4]: Leaving directory '/home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial'
make[3]: *** [Makefile:37: /home/zhangzenglin/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Error 2
make[3]: Leaving directory '/home/zhangzenglin/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#1790.37#118.95#1913.25
make[2]: *** [toolchain/Makefile:100: toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/zhangzenglin/lede'
make[1]: *** [toolchain/Makefile:96: /home/zhangzenglin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/zhangzenglin/lede'
make: *** [/home/zhangzenglin/lede/include/toplevel.mk:224：world] 错误 2

acooler15, 2020-08-05:
是用`make -j1 V=s`命令编译的吗？

zhangzenglin, 2020-08-05:
是的，我今天重新搭建了一个Ubuntu20.04的环境，编译没问题，我看来一下，估计是我的环境有问题。

closed 2021-06-20 by coolsnowwolf