coolsnowwolf/lede #5259: 求助ssr plus+访问控制失效的问题

KevinLok1872, 2020-08-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
请教一下大神们。
我是openwrt软路由lan接网件lan做AP。
然后ssr plus+开启大陆白名单模式，有些无线和有线设备IP指定全局模式。然而指定IP的设备依然使用大陆白名单模式。这是为什么呢？请问有什么办法解决呢？使用passwall也一样，是不是哪里需要别的设置？

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

yocosweet, 2020-08-06:
ssr plus+访问控制好像失效了,passwall目前没发现问题

blesswarrior, 2020-08-07:
的确是

KevinLok1872, 2020-08-07:
> ssr plus+访问控制好像失效了,passwall目前没发现问题

passwall也不行，不清楚什么情况，反而以前硬路由用梅林的时候还可以。

qiangxinglin, 2020-08-07:
是的, 我重新试验了一下, ssr+ passwall都不行

KevinLok1872, 2020-08-07:
> 是的, 我重新试验了一下, ssr+ passwall都不行

这就很麻烦。请问有别的插件推荐吗？能用的。

hcym, 2020-08-07:
没有的事，正常使用

joesdu, 2020-08-10:
我也不能正常控制,擦个眼坐等修复.

KevinLok1872, 2020-08-11:
> 我也不能正常控制，擦个眼坐等修复。

你的passwall正常吗？

joesdu, 2020-08-11:
> > 我也不能正常控制，擦个眼坐等修复。
> 
> 你的passwall正常吗？

我没有安装pass wall

KevinLok1872, 2020-08-11:
> > > 我也不能正常控制，擦个眼坐等修复。
> > 
> > 
> > 你的passwall正常吗？
> 
> 我没有安装pass wall

好吧

joesdu, 2020-08-11:
> > > > 我也不能正常控制，擦个眼坐等修复。
> > > 
> > > 
> > > 你的passwall正常吗？
> > 
> > 我没有安装pass wall
> 
> 好吧

国内IP不代理也不生效，闲鱼淘宝天天提示我别开VPN

hcym, 2020-08-11:
ssr+都正常的，passwall最近比较乱，不过前天编译的也行

joesdu, 2020-08-11:
> ssr+都正常的，passwall最近比较乱，不过前天编译的也行

我HelloWorld已经好几十天都不正常了，我明天重新编译一个看看

JustinhoCHN, 2020-08-14:
> > ssr+都正常的，passwall最近比较乱，不过前天编译的也行
> 
> 我HelloWorld已经好几十天都不正常了，我明天重新编译一个看看

你这边ssr+正常吗？3天前我重头编译过一次都是强制域名走代理失效。

hcym, 2020-08-14:
我编译的都是armv8，大雕的昨天晚上还是正常，非常正常。 passwall有一个星期都没有编译，周末九代目到货准备试试x64😂

kkfuzi, 2020-09-07:
有更新吗？ssr+的访问控制好像还是不行。。

closed 2021-06-20 by coolsnowwolf