coolsnowwolf/lede #5262: IPSec VPN连上后无法上网但能访问内网设备

zeruns, 2020-08-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：手机通过IPSec VPN连上路由器后无法上网但能访问内网设备

### (2) 路由器型号和固件版本

A：X86 
固件版本：R20.7.20

### (3) 详细日志

A：

BecodReyes, 2020-08-05:
路由表/防火墙问题
请在防火墙列表/路由表中添加相应配置
具体请自行谷歌

zeruns, 2020-08-05:
> 路由表/防火墙问题
> 请在防火墙列表/路由表中添加相应配置
> 具体请自行谷歌

从旧版升上来后才有这问题的，配置没改过，旧版没问题

mjyhj, 2020-08-05:
4.14 内核：大概率用不了
4.19 内核：大概率重启
5.4 内核：只能访问内网

zeruns, 2020-08-05:
> 4.14 内核：大概率用不了
> 4.19 内核：大概率重启
> 5.4 内核：只能访问内网

对我的就是4.19内核，有时候连上后一段时间就会重启

zeruns, 2020-08-05:
> 4.14 内核：大概率用不了
> 4.19 内核：大概率重启
> 5.4 内核：只能访问内网

有没有最稳定最少bug的版本推荐

njf1993, 2020-08-07:
我也同样问题，同5.4内核，同连接成功后只能访问内网设备，无法访问外网

dumk1217, 2020-08-08:
5.4 ，我也不能访问外网

cnzd, 2020-08-08:
我也是

iloveyam, 2020-08-25:
> 路由表/防火墙问题
> 请在防火墙列表/路由表中添加相应配置
> 具体请自行谷歌

搜不到怎么设置，请链接谢谢

lizhongit, 2020-12-08:
是能上外网的，只不过域名解析存在问题，我通过tcpdump排查发现DNS的请求包没有回复，我不太熟悉iptables，应该就是配置好它的规则让DNS的包能发送到公网DNS服务器并能接收到DNS的答复包就可以。

目前解决的方法可以线让DNS在本地解析好，解析到IP地址后就可以通过IPSec访问外网资源

1368129224, 2021-01-03:
x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问

gddfczl, 2021-01-04:
> 是能上外网的，只不过域名解析存在问题，我通过tcpdump排查发现DNS的请求包没有回复，我不太熟悉iptables，应该就是配置好它的规则让DNS的包能发送到公网DNS服务器并能接收到DNS的答复包就可以。
> 
> 目前解决的方法可以线让DNS在本地解析好，解析到IP地址后就可以通过IPSec访问外网资源

请教具体怎么做？我现在是电脑用VPN Access Manager使用IPSET连接路由器，内网的网址都可以访问，外网走UDP的都好用，例如QQ、微信都能登陆发送信息，但是网页打不开。

aepub, 2021-01-06:
遇到同样的问题，固件版本OpenWrt R20.12.12，以前老版本是可以的，寻求解决办法！

lonelysoul, 2021-01-06:
长期关注此问题，还没有遇到可行方法。多数回复都是思路加谷歌。
建议选择时一并选择其他VPN组件作为替代使用使用。例如openVPN等。
目前自己在用openVPN。可用性90%以上。

获取 Outlook for iOS<https://aka.ms/o0ukef>

940842546, 2021-01-10:
> x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问

一样 只能访问路由 其他啥都不能访问 ping都ping不通

940842546, 2021-01-10:
> > x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问
> 
> 一样 只能访问路由 其他啥都不能访问 ping都ping不通

解决了 防火墙里面把转发改成接受 就好了

940842546, 2021-01-10:
> > > x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问
> > 
> > 
> > 一样 只能访问路由 其他啥都不能访问 ping都ping不通
> 
> 解决了 防火墙里面把转发改成接受 就好了

等下 这样是只能访问内网

940842546, 2021-01-10:
> > > > x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问
> > > 
> > > 
> > > 一样 只能访问路由 其他啥都不能访问 ping都ping不通
> > 
> > 
> > 解决了 防火墙里面把转发改成接受 就好了
> 
> 等下 这样是只能访问内网

外网按照ip是能访问的 说明是dns问题

1368129224, 2021-01-11:
> > > > > x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问
> > > > 
> > > > 
> > > > 一样 只能访问路由 其他啥都不能访问 ping都ping不通
> > > 
> > > 
> > > 解决了 防火墙里面把转发改成接受 就好了
> > 
> > 
> > 等下 这样是只能访问内网
> 
> 外网按照ip是能访问的 说明是dns问题

已使用openvpn代替，Android和iOS都有官方应用(iOS需使用其他国家地区app store下载

josonwang, 2021-01-13:
现在这个还是有问题，连接好IPSEC VPN后微信可以访问，但是网页外网网址是打不开的。

yishion819, 2021-01-24:
> 
> 
> > > x86_64，5.4内核，连接后只能访问路由，其他内网外网都不能访问
> > 
> > 
> > 一样 只能访问路由 其他啥都不能访问 ping都ping不通
> 
> 解决了 防火墙里面把转发改成接受 就好了

请问你的OP版本是？ 我的版本是4.19.138 防火墙转发改为接受还是无法上网  谢谢

veoco, 2021-02-04:
iptables -I FORWARD -s 192.168.0.10/24 -j ACCEPT
iptables -t nat -I PREROUTING 2 -s 192.168.0.10/24 -p udp --dport 53 -j ACCEPT

这样国内访问没有问题，但 SSR+ 还是走不通。

yishion819, 2021-02-05:
> 
> 
> iptables -I FORWARD -s 192.168.0.10/24 -j ACCEPT
> iptables -t nat -I PREROUTING 2 -s 192.168.0.10/24 -p udp --dport 53 -j ACCEPT
> 
> 这样国内访问没有问题，但 SSR+ 还是走不通。

我这边在4.19.138上面 按照你的设置后 国内能访问了 但SSR会大概率崩溃  我现在已经换成5.4了  目前正常

veoco, 2021-02-05:
新解决方法，想要的功能基本都恢复正常了。由于我最初的报错信息与 #5184 相同，加上前面有人说不同内核版本症状不一样，导致走了些弯路，之前的方法现在看看就好了。

一、连上只能访问内网，和完全无法连上并且系统日志内报错和  #5184 相同有这么两行：

```
Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Sat Sep 19 16:19:00 2020 daemon.info : 00[KNL] failed to create TUN device
```

字面意思是没有安装 kmod-tun，导致无法创建 TUN 设备。如果在编译时勾选上 kmod-tun，或者手动安装 kmod-tun 后就没有这个报错，并且可以正常连上和访问内网，只是无法访问外网。通过我上面的两条命令虽然可以修复，但其实修复思路完全错误。

导致问题的原因其实是 https://github.com/coolsnowwolf/lede/commit/9905c8cd15079eead415fdd4302be53e8f4183ab 这个提交增加了对 strongswan-mod-kernel-libipsec 依赖，我不清楚是因为什么原因增加了这个依赖。但这样会使 strongSwan 启用 kernel-libipsec 插件，将 IPsec 流量从内核转到其自身创建的 TUN 设备。结果就是原先的 iptables 对 IPSec 流量的管理完全失效。

这个问题的解决方法就是将 luci-app-ipsec-vpnd 的 Makefile 中的 strongswan-mod-kernel-libipsec 依赖删除，重新编译安装即可，这个问题影响了 2020 年 7 月 8 日以后的所有版本，不分内核。

二、内外网可以访问，但 SSR+ 对于 IPSec 的流量无效

解决掉上一个问题后，我将 luci-app-ipsec-vpnd 的源码看了很多遍，十分疑惑为什么之前 IPSec 流量会走 SSR+，从现有代码完全看不到有这个功能，应该只有从 Lan 过来的流量和本机应用流量会走 SSR+。

直到我看到了 https://github.com/fw876/helloworld/commit/0c1873f018fc50f8f3859df6749465b13e9c7261 ，原本 SSR+ 的iptables 规则是想要限定 Lan 的，但是由于获取 Lan 名称的代码有 Bug，导致从任意接口不论 IPSec 甚至 Wan 过来的流量都会走 SSR+。而这个提交修复了这个 BUG，因此现在 SSR+ 对于 IPSec 的流量无效。

让 IPSec 按规则走 SSR+ 方法是使用：

```
iptables -t nat -I PREROUTING -p tcp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_WAN_AC
iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY
```

这问题影响了 2020 年11 月16 日以后的所有版本，可以说是基于 Bug 的功能由于 Bug 被修复而失效了。

yishion819, 2021-02-07:
> strongswan-mod-kernel-libipsec

我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包  结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢

veoco, 2021-02-07:
> 
> 
> > strongswan-mod-kernel-libipsec
> 
> 我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包 结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢

不清楚你是什么状况，这是我现在用的，你可以试试。

[ipsec.zip](https://github.com/coolsnowwolf/lede/files/5938696/ipsec.zip)

yishion819, 2021-02-07:
> 
> 
> > > strongswan-mod-kernel-libipsec
> > 
> > 
> > 我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包 结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢
> 
> 不清楚你是什么状况，这是我现在用的，你可以试试。
> 
> [ipsec.zip](https://github.com/coolsnowwolf/lede/files/5938696/ipsec.zip)

你有编译的X86的EFI启动的OP固件么? 我现在这个安装不上你这个插件

veoco, 2021-02-07:
> 
> 
> > > > strongswan-mod-kernel-libipsec
> > > 
> > > 
> > > 我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包 结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢
> > 
> > 
> > 不清楚你是什么状况，这是我现在用的，你可以试试。
> > [ipsec.zip](https://github.com/coolsnowwolf/lede/files/5938696/ipsec.zip)
> 
> 你有编译的X86的EFI启动的OP固件么? 我现在这个安装不上你这个插件

并没有，我用 R2S 的。

yishion819, 2021-02-07:
> 
> 
> > > > > strongswan-mod-kernel-libipsec
> > > > 
> > > > 
> > > > 我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包 结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢
> > > 
> > > 
> > > 不清楚你是什么状况，这是我现在用的，你可以试试。
> > > [ipsec.zip](https://github.com/coolsnowwolf/lede/files/5938696/ipsec.zip)
> > 
> > 
> > 你有编译的X86的EFI启动的OP固件么? 我现在这个安装不上你这个插件
> 
> 并没有，我用 R2S 的。

不行 我换了一个固件 可以安装你这个插件了 安装后还是无法上网

yishion819, 2021-02-09:
> 
> 
> > > > > strongswan-mod-kernel-libipsec
> > > > 
> > > > 
> > > > 我这个5.4的目前是连上VPN后 可以翻出去 国内无法访问 我搜索了软件包 结果没有strongswan-mod-kernel-libipsec这个 可否帮忙编译一个IPK文件呢？谢谢
> > > 
> > > 
> > > 不清楚你是什么状况，这是我现在用的，你可以试试。
> > > [ipsec.zip](https://github.com/coolsnowwolf/lede/files/5938696/ipsec.zip)
> > 
> > 
> > 你有编译的X86的EFI启动的OP固件么? 我现在这个安装不上你这个插件
> 
> 并没有，我用 R2S 的。

可否帮忙编译一个X86的5.4版本的strongswan-mod-kernel-libipsec呢？十分感谢

veoco, 2021-02-09:
> 可否帮忙编译一个X86的5.4版本的strongswan-mod-kernel-libipsec呢？十分感谢

strongswan-mod-kernel-libipsec 这个就是要去掉的，不应该安装这个。

bobcc4, 2021-02-11:
> 新解决方法，想要的功能基本都恢复正常了。由于我最初的报错信息与 #5184 相同，加上前面有人说不同内核版本症状不一样，导致走了些弯路，之前的方法现在看看就好了。
> 
> 一、连上只能访问内网，和完全无法连上并且系统日志内报错和 #5184 相同有这么两行：
> 
> ```
> Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
> Sat Sep 19 16:19:00 2020 daemon.info : 00[KNL] failed to create TUN device
> ```
> 
> 字面意思是没有安装 kmod-tun，导致无法创建 TUN 设备。如果在编译时勾选上 kmod-tun，或者手动安装 kmod-tun 后就没有这个报错，并且可以正常连上和访问内网，只是无法访问外网。通过我上面的两条命令虽然可以修复，但其实修复思路完全错误。
> 
> 导致问题的原因其实是 [9905c8c](https://github.com/coolsnowwolf/lede/commit/9905c8cd15079eead415fdd4302be53e8f4183ab) 这个提交增加了对 strongswan-mod-kernel-libipsec 依赖，我不清楚是因为什么原因增加了这个依赖。但这样会使 strongSwan 启用 kernel-libipsec 插件，将 IPsec 流量从内核转到其自身创建的 TUN 设备。结果就是原先的 iptables 对 IPSec 流量的管理完全失效。
> 
> 这个问题的解决方法就是将 luci-app-ipsec-vpnd 的 Makefile 中的 strongswan-mod-kernel-libipsec 依赖删除，重新编译安装即可，这个问题影响了 2020 年 7 月 8 日以后的所有版本，不分内核。
> 
> 二、内外网可以访问，但 SSR+ 对于 IPSec 的流量无效
> 
> 解决掉上一个问题后，我将 luci-app-ipsec-vpnd 的源码看了很多遍，十分疑惑为什么之前 IPSec 流量会走 SSR+，从现有代码完全看不到有这个功能，应该只有从 Lan 过来的流量和本机应用流量会走 SSR+。
> 
> 直到我看到了 [fw876/helloworld@0c1873f](https://github.com/fw876/helloworld/commit/0c1873f018fc50f8f3859df6749465b13e9c7261) ，原本 SSR+ 的iptables 规则是想要限定 Lan 的，但是由于获取 Lan 名称的代码有 Bug，导致从任意接口不论 IPSec 甚至 Wan 过来的流量都会走 SSR+。而这个提交修复了这个 BUG，因此现在 SSR+ 对于 IPSec 的流量无效。
> 
> 让 IPSec 按规则走 SSR+ 方法是使用：
> 
> ```
> iptables -t nat -I PREROUTING -p tcp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_WAN_AC
> iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY
> ```
> 
> 这问题影响了 2020 年11 月16 日以后的所有版本，可以说是基于 Bug 的功能由于 Bug 被修复而失效了。

5.4内核：按照要求把依赖删掉重新编译还是无法访问外网（如百度之类）

cn0068, 2021-02-12:
学习了一下，终于解决了访问外网，科学问题。

yishion819, 2021-02-15:
> 
> 
> 学习了一下，终于解决了访问外网，科学问题。

你是怎么解决的

hnrrn, 2021-02-18:
> strongswan-mod-kernel-libipsec 这个就是要去掉的，不应该安装这个。

最新内核5.4.95 去strongswan-mod-kernel-libipsec 连不上了

milesking, 2021-02-19:
> 最新内核5.4.95 去strongswan-mod-kernel-libipsec 连不上了

5.4.92 本来想移除strongswan-mod-kernel-libipsec，还是不作死了

cn0068, 2021-02-19:
我自己编译别人的，现在可以用了，VPN科学上网访问内网，外网都正常

bobcc4, 2021-02-19:
还是无法使用，尝试过 strongswan-mod-kernel-libipse ，退回4.19版本，都还是不行。  已转用softether

xwysun, 2021-02-19:
> […](#)

请问怎么解决的，我用了楼上的ipsec的包，也删掉了strongswan-mod-kernel-libipsec，但是问题更多了，连网关都无法访问了

veoco, 2021-02-19:
不要在原有固件直接删除 strongswan-mod-kernel-libipsec 或 luci-app-ipsec-vpnd 后再安装我上传的那个 luci-app-ipsec-vpnd。这样是修复不了的（可能会由于依赖问题多删或者少删除包）。

最好重新拉取代码，然后删除依赖，从头开始编译整个固件。确保编译出的固件没有受到之前配置的影响，再升级编译出的固件，升级时勾选不保留配置。

hnrrn, 2021-02-20:
> 从头开始编译整个固件

这个方法适合最新的内核吗 下午准备试试

closed 2021-02-21 by coolsnowwolf

yishion819, 2021-02-21:
> 
> […](#)
> -- Best regards.

可否分享你编译的固件呢？

lonelysoul, 2021-02-21:
合并了pr，现在编译应该没有这个了。新编一下看看。
获取 Outlook for iOS<https://aka.ms/o0ukef>

cn0068, 2021-02-22:
QQ: 61008008

hnrrn, 2021-02-23:
> > […](#)
> 
> 请问怎么解决的，我用了楼上的ipsec的包，也删掉了strongswan-mod-kernel-libipsec，但是问题更多了，连网关都无法访问了

我和你一样操作 结果也一样连网关都访问不到了

zeruns, 2021-02-23:
> > > […](#)
> > 
> > 
> > 请问怎么解决的，我用了楼上的ipsec的包，也删掉了strongswan-mod-kernel-libipsec，但是问题更多了，连网关都无法访问了
> 
> 我和你一样操作 结果也一样连网关都访问不到了

我重新编译了最新版的源码后可以了，功能正常

hnrrn, 2021-02-23:
> > > > […](#)
> > > 
> > > 
> > > 请问怎么解决的，我用了楼上的ipsec的包，也删掉了strongswan-mod-kernel-libipsec，但是问题更多了，连网关都无法访问了
> > 
> > 
> > 我和你一样操作 结果也一样连网关都访问不到了
> 
> 我重新编译了最新版的源码后可以了，功能正常

能发一下你的 menu config 吗。。我准备用你的配置试试

zeruns, 2021-02-23:
> > > > > […](#)
> > > > 
> > > > 
> > > > 请问怎么解决的，我用了楼上的ipsec的包，也删掉了strongswan-mod-kernel-libipsec，但是问题更多了，连网关都无法访问了
> > > 
> > > 
> > > 我和你一样操作 结果也一样连网关都访问不到了
> > 
> > 
> > 我重新编译了最新版的源码后可以了，功能正常
> 
> 能发一下你的 menu config 吗。。我准备用你的配置试试

配置文件和编译好的img文件在这：https://zeruns.lanzous.com/i1OO2m1nw1e

LeeJunwei, 2021-03-14:
> 新解决方法，想要的功能基本都恢复正常了。由于我最初的报错信息与 #5184 相同，加上前面有人说不同内核版本症状不一样，导致走了些弯路，之前的方法现在看看就好了。
>
> 一、连上只能访问内网，和完全无法连上并且系统日志内报错和 #5184 相同有这么两行：
>
> ```
> Sat Sep 19 16:19:00 2020 daemon.info : 00[LIB] failed to open [...]

运行iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY后，报：“iptables v1.8.4 (legacy): Couldn't load target `SS_SPEC_TPROXY':No such file or directory”是什么原因呢？

veoco, 2021-03-15:
> 运行iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY后，报：“iptables v1.8.4 (legacy): Couldn't load target `SS_SPEC_TPROXY':No such file or [...]

说明没开 UDP 中继服务器，这条就不需要了。

LeeJunwei, 2021-03-16:
> 说明没开 UDP 中继服务器，这条就不需要了。

但是只运行那三条指令，手机连上IPSec后，还是不能走SSR

eearphon, 2021-06-09:
> > > 
> > > 运行iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY后，报：“iptables v1.8.4 (legacy): Couldn't load target `SS_SPEC_TPROXY':No such file or directory”是什么原因呢？
> > 
> > 
> > 说明没开 UDP 中继服务器，这条就不需要了。
> 
> 但是只运行那三条指令，手机连上IPSec后，还是不能走SSR

同样问题，请问您解决问题了吗？

LeeJunwei, 2021-06-09:
没有解决

发自我的iPad

veoco, 2021-06-09:
> 
> 
> > > > 运行iptables -t mangle -I PREROUTING -p udp -m policy --dir in --pol ipsec --proto esp -j SS_SPEC_TPROXY后，报：“iptables v1.8.4 (legacy): Couldn't load target `SS_SPEC_TPROXY':No such file or directory”是什么原因呢？
> > > 
> > > 
> > > 说明没开 UDP 中继服务器，这条就不需要了。
> > 
> > 
> > 但是只运行那三条指令，手机连上IPSec后，还是不能走SSR
> 
> 同样问题，请问您解决问题了吗？

当初复现不了这个问题，我现在也不折腾 LEDE 了。如果这边 ipsec 能正常跑，但是 SSR+ 走不了，建议去 https://github.com/fw876/helloworld/ 开 issue。

eearphon, 2021-06-10:
重新默认编译了x86_64版本，最新 d871216，能连上，但啥网也访问不了，加上上方那些命令也不行

hu-zhi-hang, 2021-06-10:
> 重新默认编译了x86_64版本，最新 [d871216](https://github.com/coolsnowwolf/lede/commit/d871216a7ce1f1a6b8efae99870361dfd74b7ff8)，能连上，但啥网也访问不了，加上上方那些命令也不行

折腾几次我已经放弃用 ipsec 了，问题很多，openvpn 稳定一点

lonelysoul, 2021-07-04:
转载：
亲自验证在接口那添加一个接口即可实现外网，然后在ipsec里面客户端地址段设置为你的内网网段，比如192.168.1.250/24，dns填写内网网关，比如192.168.1.1； 开启ipsec，在网络-接口里添加新接口，静态地址，IP地址随便填一个不是你内网网段的就可以，比如10.10.100.1,子网掩码随便填，比如255.255.255.100;其它不用管；物理设置里选择ipsec0接口（如果没有这个端口建议更换一个版本最省事），保存后到网络-防火墙基本设置里面，转发改为接受即可
最后重启路由就可以愉快使用IPSec的转发了

PR #6331 (2021-02-21): Revert "luci-app-ipsec-vpnd: add strongswan-mod-kernel-libipsec"
Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道

Fixes #5262 and #5184