coolsnowwolf/lede #5264: 关于x86软路由docker-ce无法运行

kisaragychihaya, 2020-08-05:
系统是x86 根据docker-ce官方的[包](https://github.com/openwrt/packages/blob/master/utils/docker-ce/files/etc/config/dockerd) 
把空余空间挂载的地方应该是/opt/docker 如果需要用/opt运行需要添加配置文件
```
root@OpenWrt:/# cat /etc/docker/daemon.json 
{
"data-root": "/opt/",
"log-level": "warn"
}
```
否则就会有如下错误
```
docker: Error response from daemon: OCI runtime create failed: /var/lib/docker/overlay2/1f4e00aed98082a6b80024cd05efed9eea54754975bd781b43741833a0ee1a29/merged is not an absolute path or is a symlink: unknown
```
望可以把这个配置文件整合进默认配置里或者把创建配置文件的步骤写进教程

mjyhj, 2020-08-05:
自己添加PR，或者自己files编译进去

darkalexx, 2020-08-06:
大佬能不能说一下具体的方法，我试了把挂载点设置为/opt/docker 还是一样的报错

kisaragychihaya, 2020-08-06:
PR[#5272 ](https://github.com/coolsnowwolf/lede/pull/5272)

closed 2020-08-06 by kisaragychihaya