coolsnowwolf/lede #5270: 最新版本编译错误，exfat冲突

Cathgao, 2020-08-06:
在#5249 更新之后，旧的配置文件中出现exfat冲突导致编译失败
```
2020-08-05T23:05:35.2876719Z  * check_data_file_clashes: Package kmod-fs-exfat wants to install file /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/modules/4.19.131/exfat.ko
2020-08-05T23:05:35.2879250Z Configuring kmod-ath10k-ct.
2020-08-05T23:05:35.2881520Z 	But that file is already provided by package  * kmod-fs-exfat0
2020-08-05T23:05:35.2881820Z Configuring kmod-nls-cp936.
2020-08-05T23:05:35.2883071Z  * opkg_install_cmd: Cannot install package kmod-fs-exfat.
2020-08-05T23:05:35.2883373Z Configuring luci-i18n-base-zh-cn.
2020-08-05T23:05:35.2884687Z  * check_data_file_clashes: Package kmod-fs-exfat wants to install file /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/lib/modules/4.19.131/exfat.ko
2020-08-05T23:05:35.2884886Z Configuring fdisk.
2020-08-05T23:05:35.2885922Z 	But that file is already provided by package  * kmod-fs-exfat0
2020-08-05T23:05:35.2886207Z Configuring ip6tables-extra.
2020-08-05T23:05:35.2890433Z  * opkg_install_cmd: Cannot install package automount.
```
请问`kmod-fs-exfat0`与`kmod-fs-exfat`谁才是更新的？

minlang112, 2020-08-06:
请查看更新详细内容 https://github.com/coolsnowwolf/lede/pull/5249/files

kmod-fs-exfat0是旧的，kmod-fs-exfat是新的

AmadeusGhost, 2020-08-06:
make clean 后重新编译即可

yangabc123, 2020-08-06:
make clean后还是不行

Cathgao, 2020-08-06:
> make clean后还是不行

不是make clean的问题，autosamba选中的是新的exfat，如果自己又选中一个，就会冲突，需要修改配置

Cathgao, 2020-08-06:
取消选中kmod-fs-exfat0后编译通过

closed 2020-08-06 by Cathgao