coolsnowwolf/lede #5281: 您当前服务器不支持PATH_INFO模式

Gokou-Ruri, 2020-08-06:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
#4813 

## 2. 详细叙述

### (1) 具体问题

A：在Luci上设置好可道云，更新成功后进入可道云，配置WebDAV插件的时候弹出提示。
[image]

### (2) 路由器型号和固件版本

| 主机型号 |	Newifi-D2 |
| ------------ | ------------ |
| 架构	| MediaTek MT7621 ver:1 eco:3 x 4 |
| 固件版本 |	OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae) |
| 内核版本 |	5.4.51 |

### (3) 详细日志

A：
| 存储设备路径 | /mnt/sda1/ |
| :------------ | :------------ |
| 项目存放目录 | /mnt/sda1/kodbox |

EnnawYang, 2020-08-07:
https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-kodexplorer 替换下即可

Gokou-Ruri, 2020-08-08:
> https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-kodexplorer 替换下即可
> [xiaorouji/openwrt-package#453](https://github.com/xiaorouji/openwrt-package/issues/453)

谢谢大佬！请问这个MIPS路由器能用吗？是不是编译出ipk直接覆盖安装就行？

Gokou-Ruri, 2020-08-09:
> https://github.com/xiaorouji/openwrt-package/tree/master/lienol/luci-app-kodexplorer 替换下即可
> [xiaorouji/openwrt-package#453](https://github.com/xiaorouji/openwrt-package/issues/453)

谢谢大佬，替换源码编译安装后可以用了，不过不能上传，只能下载，还是不用kodbox了。

closed 2020-08-09 by Gokou-Ruri