coolsnowwolf/lede #5282: 关于ssr plus+负载均衡的疑问

cn10010, 2020-08-07:
[反馈bug/问题模板，提建议请删除]()

## 关于ssr plus+负载均衡的疑问

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用haproxy来做负载均衡。用的是v2ray，刚开始是可以用的，但是过几分钟看网络连接状态v2ray有一堆的tcp连接，看haproxy日志是ulimit值超出？但是我根本没有打开几个网页，也没有大流量的上网。
[image]
查看haproxy网页这个是什么操作
[image]
尝试过各种改ulimit值，增加ip，端口复用也都没有用，但是重启haproxy之后又能使用几分钟，之后重复上面的问题
我应该怎么做才能解决这个问题

### (1) 具体问题

A：使用haproxy搭配ssr plus产生会产生大量tcp连接导致haproxy出错

### (2) 路由器型号和固件版本

A： x86 

### (3) 详细日志

A：

cn10010, 2020-08-07:
如果单单使用ssr plus+不经过haproxy就没有问题，是我配置有问题吗，在网上找了一圈也没有解决

hcym, 2020-08-07:
只能用gfwlist模式，路由白名单的不能用，而且非常容易卡死，官方的和lion没这个问题。其他的都没有问题

cn10010, 2020-08-07:
试了一下，改成gfw模式确实不会有这样的问题，如果想要用白名单模式还有希望吗

cn10010, 2020-08-07:
> 只能用gfwlist模式，路由白名单的不能用，而且非常容易卡死，官方的和lion没这个问题。其他的都没有问题

试了一下，改成gfw模式确实不会有这样的问题，如果想要用白名单模式还有希望吗

hcym, 2020-08-07:
这个问题只能等大雕，我是无所谓的，一直用U盘系统，各种配方偏方换个盘而已😂

coolsnowwolf, 2020-08-14:
OpenWrt Linux 内核默认文件描述符太少，SSR Plus 我做了加大，HAPROXY 你可以参考修改

hcym, 2020-08-14:
完全不懂啊大佬，list正常使用就行了，不过haproxy新版本对多核心的支持更好了，luci-app要是能够更新一下就好啦🤣

cn10010, 2020-09-01:
> OpenWrt Linux 内核默认文件描述符太少，SSR Plus 我做了加大，HAPROXY 你可以参考修改

可以直接改linux的吗

hcym, 2020-09-17:
用官方19.07.4编译的可以啦，不怕稳定性问题的话用官方master分支也行。就是启动ssr时cpu会比较高，小机器可能吃不消。但是会慢慢恢复平静😂

cn10010, 2020-09-17:
> 用官方19.07.4编译的可以啦，不怕稳定性问题的话用官方master分支也行。就是启动ssr时cpu会比较高，小机器可能吃不消。但是会慢慢恢复平静😂

感觉cpu占用不成问题，毕竟除了用来扶墙软路由好像也没有其他用途。

cn10010, 2020-09-17:
现在用脚本来判断访问谷歌状态，访问不了就重启haproxy。暂时用这个脚本苟活一下😂

closed 2021-06-20 by coolsnowwolf

cary-sas, 2021-09-23:
> 反馈bug/问题模板，提建议请删除
>
> ## 关于ssr plus+负载均衡的疑问
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述 [...]

可以参考一下haproxy 把 v2ray节点做负载平衡的配置文件吗