coolsnowwolf/lede #5284: 编译好的固件写盘到97.8%时 Error#0 occured while writting to disk at sector XXXXX

seasun2k, 2020-08-07:
编译了几次都是这样......写其他人的固件就没有问题.....？？？？？

hisune, 2020-08-07:
遇到过，换DiskImage1.2版本写入就可以了

Twinzo1, 2020-08-07:
换个写盘工具再试试，比如Etcher

smalldickman, 2020-08-08:
有没有修改kernel和root区大小阿(Target Images  --->)，修改会容易出现这种现象，我以前改的128mb+896mb就写盘错误，后来改到16mb+512mb就没问题了..。。。。.原因的一种

seasun2k, 2020-08-08:
> 有没有修改kernel和root区大小阿(Target Images --->)，修改会容易出现这种现象，我以前改的128mb+896mb就写盘错误，后来改到16mb+512mb就没问题了..。。。。.原因的一种

没有修改过，默认值。感觉应该是配置的问题....

seasun2k, 2020-08-08:
> 遇到过，换DiskImage1.2版本写入就可以了

别人编译的估计就能正确写入，感觉还是配置的问题

xiasonming, 2020-08-09:
> > 遇到过，换DiskImage1.2版本写入就可以了
> 
> 别人编译的估计就能正确写入，感觉还是配置的问题
兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？

xiasonming, 2020-08-09:
兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？

seasun2k, 2020-08-09:
> 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？

没有，试了好多次，就是在写固件到最后的时候报错。:(

smalldickman, 2020-08-10:
> > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> 
> 没有，试了好多次，就是在写固件到最后的时候报错。

> > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> 
> 没有，试了好多次，就是在写固件到最后的时候报错。:(

留个邮箱，我把我用的包括写盘工具发你，测试下

seasun2k, 2020-08-10:
> > > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> > 
> > 
> > 没有，试了好多次，就是在写固件到最后的时候报错。
> > 、
> > 留个邮箱，我把我用的包括写盘工具发你，测试下，可以写到闲置的空U盘试一下不报错再正式安装（要是没有修改默认大小的话，是不是选的东西有点多，16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs）

3296458@qq.com，我试试，多谢兄弟。

dingpengyu, 2020-08-10:
> > > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> > 
> > 
> > 没有，试了好多次，就是在写固件到最后的时候报错。
> > 、
> > 留个邮箱，我把我用的包括写盘工具发你，测试下，可以写到闲置的空U盘试一下不报错再正式安装（要是没有修改默认大小的话，是不是选的东西有点多，16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs）

能否也给我发一套？我的邮箱：
1650649574@qq.com

dingpengyu, 2020-08-10:
我I36100U的路由刷机也会遇到此问题。

smalldickman, 2020-08-10:
> > > > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> > > 
> > > 
> > > 没有，试了好多次，就是在写固件到最后的时候报错。
> > > 、
> > > 留个邮箱，我把我用的包括写盘工具发你，测试下，可以写到闲置的空U盘试一下不报错再正式安装（要是没有修改默认大小的话，是不是选的东西有点多，16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs）
> 
> 能否也给我发一套？我的邮箱：
> [1650649574@qq.com](mailto:1650649574@qq.com)

以发

seasun2k, 2020-08-13:
> > > > > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> > > > 
> > > > 
> > > > 没有，试了好多次，就是在写固件到最后的时候报错。
> > > > 、
> > > > 留个邮箱，我把我用的包括写盘工具发你，测试下，可以写到闲置的空U盘试一下不报错再正式安装（要是没有修改默认大小的话，是不是选的东西有点多，16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs）
> > 
> > 
> > 能否也给我发一套？我的邮箱：
> > [1650649574@qq.com](mailto:1650649574@qq.com)
> 
> 两位已发

用你的img文件写盘没有问题.....OK，【16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs】这个也试过了，还是报Error#0 occured while writting to disk at sector XXXXX，都是在写到97%左右的时候报错。......能否把您的.config文件给我发一份我再编译一次试试？3296458@qq.com   ...........到目前为止我所有编译出来的固件都是报同样这个问题.... :-(

smalldickman, 2020-08-13:
> > > > > > 兄弟，你的问题解决了吗？我碰到的问题和你一样，也一直不知怎么解决，难道是源码默认配置问题？
> > > > > 
> > > > > 
> > > > > 没有，试了好多次，就是在写固件到最后的时候报错。
> > > > > 、
> > > > > 留个邮箱，我把我用的包括写盘工具发你，测试下，可以写到闲置的空U盘试一下不报错再正式安装（要是没有修改默认大小的话，是不是选的东西有点多，16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs）
> > > 
> > > 
> > > 能否也给我发一套？我的邮箱：
> > > [1650649574@qq.com](mailto:1650649574@qq.com)
> > 
> > 
> > 两位已发
> 
> 用你的img文件写盘没有问题.....OK，【16mb那个别动，改下下边那个256MB或者512MB，格式选squashfs】这个也试过了，还是报Error#0 occured while writting to disk at sector XXXXX，都是在写到97%左右的时候报错。......能否把您的.config文件给我发一份我再编译一次试试？[3296458@qq.com](mailto:3296458@qq.com) ...........到目前为止我所有编译出来的固件都是报同样这个问题.... :-(

已发

seasun2k, 2020-09-23:
哪位大神指导下，我所有编译出来的版本，squashfs-combined都写到最后时写盘失败，rootfs版本写盘没有问题....
到底是什么原因呢

hypnuzzz, 2020-10-01:
> 到底是什么原因呢

我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。

seasun2k, 2020-10-08:
> > 到底是什么原因呢
> 
> 我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。

写盘失败问题试了多种方法还是没有解决...写到最后虽然失败是可以用的，不过就是找不到失败的原因。另外，我试过用linux的dd写盘可以成功。

csbiao, 2020-10-13:
> > > 到底是什么原因呢
> > 
> > 
> > 我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。
> 
> 写盘失败问题试了多种方法还是没有解决...写到最后虽然失败是可以用的，不过就是找不到失败的原因。另外，我试过用linux的dd写盘可以成功。

我的也是，是不是虚拟机编译的问题？失败，但能用

seasun2k, 2020-10-22:
> > > > 到底是什么原因呢
> > > 
> > > 
> > > 我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。
> > 
> > 
> > 写盘失败问题试了多种方法还是没有解决...写到最后虽然失败是可以用的，不过就是找不到失败的原因。另外，我试过用linux的dd写盘可以成功。
> 
> 我的也是，是不是虚拟机编译的问题？失败，但能用

不是虚拟机的问题，实体机也试过，一样写入失败。最近重新更新了最新的代码，竟然写入成功了。估计是原来版本source的问题。

csbiao, 2020-10-23:
> > > > > 到底是什么原因呢
> > > > 
> > > > 
> > > > 我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。
> > > 
> > > 
> > > 写盘失败问题试了多种方法还是没有解决...写到最后虽然失败是可以用的，不过就是找不到失败的原因。另外，我试过用linux的dd写盘可以成功。
> > 
> > 
> > 我的也是，是不是虚拟机编译的问题？失败，但能用
> 
> 不是虚拟机的问题，实体机也试过，一样写入失败。最近重新更新了最新的代码，竟然写入成功了。估计是原来版本source的问题。

更新有效吗？

seasun2k, 2020-11-03:
> > > > > > 到底是什么原因呢
> > > > > 
> > > > > 
> > > > > 我也是提示写盘失败，但是居然可以用，不过我用的功能比较少，暂时还没有发现问题，也不知道有没有什么隐患。
> > > > 
> > > > 
> > > > 写盘失败问题试了多种方法还是没有解决...写到最后虽然失败是可以用的，不过就是找不到失败的原因。另外，我试过用linux的dd写盘可以成功。
> > > 
> > > 
> > > 我的也是，是不是虚拟机编译的问题？失败，但能用
> > 
> > 
> > 不是虚拟机的问题，实体机也试过，一样写入失败。最近重新更新了最新的代码，竟然写入成功了。估计是原来版本source的问题。
> 
> 更新有效吗？

有效，更新源码后，没有再出现写盘的问题了。

csbiao, 2020-11-04:
是删除后再git吗

seasun2k, 2020-11-05:
> […](#)

是的，我是全部删除了以后又git的

csbiao, 2020-11-09:
我试过，还是出错，不知道什么原因

seasun2k, 2020-11-11:
你把所有代码全部删除，重新下载最新代码，改ROM大小为300M，app只选必须。再试试。

gnou, 2020-11-25:
我也遇到了类似的问题，换了个写盘工具解决的，感谢 @Twinzo1 。 工具正如 @Twinzo1 在上面提到的，用的 Etcher，到官网下载的 “绿色版”，和 DiskImage 一样的用法

feheadls, 2020-12-16:
Etcher和rufus只能写到U盘、TF卡、移动盘上，不能写本地盘。

feheadls, 2020-12-19:
我完全重新编译了，还是报错，默认配置都报错。头大了。

EnosXu, 2021-01-03:
https://www.youtube.com/watch?v=Er36SftEsQw&t=616s
用cmd克隆完美解决

csbiao, 2021-02-28:
依然没有解决，有人知道什么原因吗？

HUiTHEME, 2021-03-07:
写进U盘一点问题都没有，就是写进硬盘，试过几个工具了，都报错。

csbiao, 2021-03-07:
U盘一样报错，但能用

feheadls, 2021-03-08:
用DiskGenius专业版的扇区复制功能写盘，就不会报错了，我是这么解决的。

TioaChan, 2021-03-09:
> 用DiskGenius专业版的扇区复制功能写盘，就不会报错了，我是这么解决的。

同样的问题，使用DiskGenius扇区复制成功写入，感谢

closed 2021-06-20 by coolsnowwolf