coolsnowwolf/lede #5290: 编译出来的固件应该怎么选择

xiasonming, 2020-08-08:
我刚刚开始接触软路由，以下是我编译的结果：
*openwrt-x86-64-generic-kernel.bin
*openwrt-x86-64-generic-squashfs-combined-efi.img
*openwrt-x86-64-generic-squashfs-combined-efi.vmdk
*openwrt-x86-64-generic-squashfs-combined.img
*openwrt-x86-64-generic-squashfs-combined.vmdk
*openwrt-x86-64-generic-squashfs-rootfs.img

请教这些文件应该怎么选择，我的软路由刚到手，硬盘是空白的。期待各位前辈的赐教！

EnnawYang, 2020-08-08:
BIOS启动选择`openwrt-x86-64-generic-squashfs-combined.img`刷入
UEFI启动选择`openwrt-x86-64-generic-squashfs-combined-efi.img`刷入

xiasonming, 2020-08-08:
> BIOS启动选择`openwrt-x86-64-generic-squashfs-combined.img`刷入
> UEFI启动选择`openwrt-x86-64-generic-squashfs-combined-efi.img`刷入

谢谢大佬的回答！其他的几个又是干什么用的呢？

EnnawYang, 2020-08-08:
> > BIOS启动选择`openwrt-x86-64-generic-squashfs-combined.img`刷入
> > UEFI启动选择`openwrt-x86-64-generic-squashfs-combined-efi.img`刷入
> 
> 谢谢大佬的回答！其他的几个又是干什么用的呢？

`rootfs.img`不带引导，`.vmdk`是虚拟机使用的，那个`kernel.bin`我不知道是干啥用的（

xiasonming, 2020-08-08:
> > > BIOS启动选择`openwrt-x86-64-generic-squashfs-combined.img`刷入
> > > UEFI启动选择`openwrt-x86-64-generic-squashfs-combined-efi.img`刷入
> > 
> > 
> > 谢谢大佬的回答！其他的几个又是干什么用的呢？
> 
> `rootfs.img`不带引导，`.vmdk`是给虚拟机用的，那个`kernel.bin`我不知道是干啥用的（

也就是说以后网页上升级刷就选择rootfs.img， 我现在用pe+DiskImage就选择squashfs-combined.img，是吗？

EnnawYang, 2020-08-08:
网页升级也用`squashfs-combined.img`

xiasonming, 2020-08-08:
谢谢你的耐心解答！

closed 2020-08-09 by xiasonming