coolsnowwolf/lede #5293: 更新到最新源码无线驱动编译出错

panther706, 2020-08-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法通过编译

### (2) 路由器型号和固件版本

A：WNDR4300/OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.o
/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.c: In function 'ieee80211_rx_napi':
/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.c:4777:3: error: implicit declaration of function 'netif_receive_skb_list'; did you mean 'netif_receive_skb'? [-Werror=implicit-function-declaration]
   netif_receive_skb_list(&list);
   ^~~~~~~~~~~~~~~~~~~~~~
   netif_receive_skb
/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.c:4782:3: error: implicit declaration of function 'skb_list_del_init'; did you mean 'hlist_del_init'? [-Werror=implicit-function-declaration]
   skb_list_del_init(skb);
   ^~~~~~~~~~~~~~~~~
   hlist_del_init
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target '/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.o' failed
make[9]: *** [/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211/rx.o] Error 1
scripts/Makefile.build:585: recipe for target '/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211' failed
make[8]: *** [/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/net/mac80211] Error 2
Makefile:1550: recipe for target '_module_/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1' failed
make[7]: *** [_module_/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1] Error 2
Makefile.build:13: recipe for target 'modules' failed
make[6]: *** [modules] Error 2
Makefile.real:93: recipe for target 'modules' failed
make[5]: *** [modules] Error 2
Makefile:120: recipe for target 'modules' failed
make[4]: *** [modules] Error 2
make[4]: Leaving directory '/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1'
Makefile:553: recipe for target '/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/.built' failed
make[3]: *** [/home/pwh/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_nand/backports-5.8-rc2-1/.built] Error 2
make[3]: Leaving directory '/home/pwh/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#4.10#1.81#6.84
package/Makefile:111: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/pwh/lede'
package/Makefile:107: recipe for target '/home/pwh/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/pwh/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pwh/lede'
/home/pwh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

xxp2277, 2020-08-08:
同样编译出错，ar71xx平台，wndr3800，github自动编译也出错，日志见https://github.com/xxp2277/WNDR3800/runs/961143248?check_suite_focus=true

AmadeusGhost, 2020-08-08:
ar71xx啊，晚上修下

PR #5295 (2020-08-09): mac80211-5.8: fix build on kernel 4.14 and sync upstream commit
fix #5293
closed 2020-08-09 by coolsnowwolf