coolsnowwolf/lede #5297: 每次重启openwrt后nginx和netdata都会报错，nginx提示80端口被占用，netdata提示Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'

gww014, 2020-08-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [y ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每次重启openwrt后nginx和netdata都会报错，nginx提示80端口被占用，netdata提示Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'

### (2) 路由器型号和固件版本

A：OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：Sun Aug  9 14:34:05 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Sun Aug  9 14:34:05 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 192.168.1.1:80 failed (98: Address in use)
Sun Aug  9 14:34:05 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Sun Aug  9 14:34:05 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 192.168.1.1:80 failed (98: Address in use)
Sun Aug  9 14:34:06 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Sun Aug  9 14:34:06 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 192.168.1.1:80 failed (98: Address in use)
Sun Aug  9 14:34:06 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Sun Aug  9 14:34:06 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 192.168.1.1:80 failed (98: Address in use)
Sun Aug  9 14:34:07 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Sun Aug  9 14:34:07 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: bind() to 192.168.1.1:80 failed (98: Address in use)
Sun Aug  9 14:34:07 2020 daemon.err nginx[12649]: 2020/08/09 14:34:05 [emerg] 12649#0: still could not bind()
Sun Aug  9 14:34:07 2020 daemon.info procd: Instance nginx::instance1 s in a crash loop 6 crashes, 2 seconds since last crash
Sun Aug  9 14:34:09 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Sun Aug  9 14:34:09 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Sun Aug  9 14:34:09 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Sun Aug  9 14:34:15 2020 user.err ddns-scripts[10582]: myddns_ipv4: GNU Wget Error: '4'
Sun Aug  9 14:34:15 2020 user.warn ddns-scripts[10582]: myddns_ipv4: Transfer failed - retry 1/0 in 60 seconds
Sun Aug  9 14:34:29 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Sun Aug  9 14:34:29 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Sun Aug  9 14:34:29 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Sun Aug  9 14:34:50 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Sun Aug  9 14:34:50 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Sun Aug  9 14:34:50 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Sun Aug  9 14:35:10 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Sun Aug  9 14:35:10 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Sun Aug  9 14:35:10 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Sun Aug  9 14:35:30 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.

eallion, 2020-08-11:
`service uhttpd disable` 试试看。

gww014, 2020-08-11:
> `service uhttpd disable` 试试看。

禁用之后岂不是网页端都废了……

eallion, 2020-08-11:
> > `service uhttpd disable` 试试看。
> 
> 禁用之后岂不是网页端都废了……

你不是安装了 nginx 吗

gww014, 2020-08-11:
> > > `service uhttpd disable` 试试看。
> > 
> > 
> > 禁用之后岂不是网页端都废了……
> 
> 你不是安装了 nginx 吗

nginx是编译的固件自带的，uhttpd禁用后nignx是没报错了，但路由器管理页面却没了。

Tue Aug 11 20:49:47 2020 cron.err crond[6657]: crond (busybox 1.31.1) started, log level 5
Tue Aug 11 20:49:48 2020 cron.err crond[6799]: crond (busybox 1.31.1) started, log level 5
Tue Aug 11 20:49:48 2020 daemon.err miniupnpd[6965]: could not open lease file: /var/upnp.leases
Tue Aug 11 20:49:48 2020 daemon.notice miniupnpd[6965]: HTTP listening on port 5000
Tue Aug 11 20:49:48 2020 daemon.notice miniupnpd[6965]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Aug 11 20:49:57 2020 daemon.info frps: Instance "main" disabled.
Tue Aug 11 20:49:57 2020 daemon.notice procd: /etc/rc.d/S99frps: [info] Instance "main" disabled.
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]: ttyd 1.6.0-f9e821f97 (libwebsockets 3.2.0)
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]: tty configuration:
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]:   start command: /usr/libexec/login.sh
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]:   close signal: SIGHUP (1)
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]:   terminal type: xterm-256color
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]:  Using foreign event loop...
Tue Aug 11 20:49:57 2020 daemon.notice ttyd[7166]:  Listening on port: 7681
Tue Aug 11 20:49:57 2020 user.info wsdd2[7250]: starting.
Tue Aug 11 20:49:57 2020 daemon.info procd: - init complete -
Tue Aug 11 20:50:00 2020 authpriv.info dropbear[7501]: Child connection from 192.168.1.17:55467
Tue Aug 11 20:50:01 2020 authpriv.notice dropbear[7501]: Password auth succeeded for 'root' from 192.168.1.17:55467
Tue Aug 11 20:50:01 2020 authpriv.info dropbear[7502]: Child connection from 192.168.1.17:55469
Tue Aug 11 20:50:01 2020 authpriv.notice dropbear[7502]: Password auth succeeded for 'root' from 192.168.1.17:55469
Tue Aug 11 20:50:09 2020 user.notice firewall: Reloading firewall due to ifup of wan (eth0)
Tue Aug 11 20:50:09 2020 cron.err crond[7853]: crond (busybox 1.31.1) started, log level 5
Tue Aug 11 20:50:10 2020 cron.err crond[7969]: crond (busybox 1.31.1) started, log level 5
Tue Aug 11 20:50:10 2020 daemon.notice miniupnpd[6965]: shutting down MiniUPnPd
Tue Aug 11 20:50:10 2020 kern.info kernel: [   51.896026] hv_balloon: Max. dynamic memory size: 2048 MB
Tue Aug 11 20:50:10 2020 daemon.err miniupnpd[8136]: could not open lease file: /var/upnp.leases
Tue Aug 11 20:50:10 2020 daemon.notice miniupnpd[8136]: HTTP listening on port 5000
Tue Aug 11 20:50:10 2020 daemon.notice miniupnpd[8136]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Aug 11 20:50:10 2020 user.notice ddns-scripts[8178]: myddns_ipv4: PID '8178' started at 2020-08-11 20:50
Tue Aug 11 20:52:15 2020 daemon.err nginx[3256]: 2020/08/11 20:52:15 [error] 3301#0: *1 open() "/www/favicon.ico" failed (2: No such file or directory), client: 192.168.1.17, server: _lan, request: "GET /favicon.ico HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/"
Tue Aug 11 20:54:50 2020 daemon.err nginx[3256]: 2020/08/11 20:54:50 [error] 3301#0: *3 open() "/www/favicon.ico" failed (2: No such file or directory), client: 192.168.1.17, server: _lan, request: "GET /favicon.ico HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/"

eallion, 2020-08-11:
`service uhttpd enable`
`service uhttpd start`

eallion, 2020-08-11:
“重新编译一般能解决90%的问题。”

Promix953, 2020-08-11:
要么nginx改监听端口
要么uhttpd改监听端口
要么不使用uhttpd，改为使用nginx提供luci页面

gww014, 2020-08-11:
> “重新编译一般能解决90%的问题。”

重新编译之后覆盖安装过好几次还是一样的问题，可能是我懒没有不保留配置，自从升级内核之后一两天内OP就会自动重启。

> 要么nginx改监听端口
> 要么uhttpd改监听端口
> 要么不使用uhttpd，改为使用nginx提供luci页面

谢谢，改端口好像很麻烦，我还是改天先重刷一次试试

Promix953, 2020-08-12:
> > “重新编译一般能解决90%的问题。”
> 
> 重新编译之后覆盖安装过好几次还是一样的问题，可能是我懒没有不保留配置，自从升级内核之后一两天内OP就会自动重启。
> 
> > 要么nginx改监听端口
> > 要么uhttpd改监听端口
> > 要么不使用uhttpd，改为使用nginx提供luci页面
> 
> 谢谢，改端口好像很麻烦，我还是改天先重刷一次试试

可以搜索下“更换openwrt默认的web server为nginx”，使用nginx提供luci的web服务，个人认为是比较好的解决办法

wushenghn, 2020-08-13:
我就用nginx正常的,选了用nginx就没有uhttp了

gww014, 2020-08-13:
> > > “重新编译一般能解决90%的问题。”
> > 
> > 
> > 重新编译之后覆盖安装过好几次还是一样的问题，可能是我懒没有不保留配置，自从升级内核之后一两天内OP就会自动重启。
> > > 要么nginx改监听端口
> > > 要么uhttpd改监听端口
> > > 要么不使用uhttpd，改为使用nginx提供luci页面
> > 
> > 
> > 谢谢，改端口好像很麻烦，我还是改天先重刷一次试试
> 
> 可以搜索下“更换openwrt默认的web server为nginx”，使用nginx提供luci的web服务，个人认为是比较好的解决办法

看过了，最大的问题要重新编译，那我还不如不保留配置重刷固件

> 我就用nginx正常的,选了用nginx就没有uhttp了

问题是我没选过，就更新了固件后两者冲突了

qq1316237299, 2020-09-10:
> > > “重新编译一般能解决90%的问题。”
> > 
> > 
> > 重新编译之后覆盖安装过好几次还是一样的问题，可能是我懒没有不保留配置，自从升级内核之后一两天内OP就会自动重启。
> > > 要么nginx改监听端口
> > > 要么uhttpd改监听端口
> > > 要么不使用uhttpd，改为使用nginx提供luci页面
> > 
> > 
> > 谢谢，改端口好像很麻烦，我还是改天先重刷一次试试
> 
> 可以搜索下“更换openwrt默认的web server为nginx”，使用nginx提供luci的web服务，个人认为是比较好的解决办法

是会重启的

Fyred, 2020-11-07:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [y ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：每次重启openwrt后nginx和netdata都会报错，nginx提示80端口被占用，netdata提示Cannot read process groups [...]

请问netdata报错Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'的问题解决了吗？

liwenjie119, 2020-12-19:
通过 /etc/netdata/edit-config apps_groups.conf 命令创建该文件。源自
https://learn.netdata.cloud/docs/agent/collectors/apps.plugin/
但是我没太看懂什么作用

Ycong, 2021-01-11:
> 通过 /etc/netdata/edit-config apps_groups.conf 命令创建该文件。源自
> https://learn.netdata.cloud/docs/agent/collectors/apps.plugin/
> 但是我没太看懂什么作用

执行
```shell
/etc/netdata/edit-config apps_groups.conf
```
后进行了一次复制操作

```shell
Copying '/usr/lib/netdata/conf.d/apps_groups.conf' to '/etc/netdata/apps_groups.conf' ... 
```

closed 2021-06-20 by coolsnowwolf