coolsnowwolf/lede #5299: 大麦DW33D编译失败

wifiway520, 2020-08-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
一直使用大雕lede仓库编译DW33D固件，也一直正常，最近发现，不管如何选择插件，或者不选，都会报too  big，随后尝试选择大雕openwrt17.01仓库，还是报错，随后又尝试修改17.01内核4.9和4.14，还是不行
### (1) 具体问题

A：lede仓库最新版只有4.14内核，是不是已经不支持大麦DW33D？openwrt-17.01里可以选择内核(4.9和4.14)但还是编译失败，我该如何操作

### (2) 路由器型号和固件版本

A：大麦DW33D。。使用固件是大雕lede–4.14内核固件

### (3) 详细日志

A：

Aes64X, 2020-08-09:
换机器

wifiway520, 2020-08-09:
我是来反馈问题和寻找办法的！楼上，你可以洗洗了

wifiway520, 2020-08-09:
> 换机器

我是来反馈问题和寻找办法的！你可以洗洗了

EnnawYang, 2020-08-10:
换机器

AmadeusGhost, 2020-08-10:
> 我是来反馈问题和寻找办法的！你可以洗洗了

会改代码就自己做喽，这个问题是 kernel 分区过小所致的
dw33d 我的在学校，所以我只能下个月才能修下这个

wifiway520, 2020-08-10:
> > 我是来反馈问题和寻找办法的！你可以洗洗了
> 
> 会改代码就自己做喽，这个问题是 kernel 分区过小所致的
> dw33d 我的在学校，所以我只能下个月才能修下这个

大佬，我只是小白类，不会代码，所以只能反馈，希望能得大佬们的帮助

Aes64X, 2020-08-10:
> > > 我是来反馈问题和寻找办法的！你可以洗洗了
> > 
> > 
> > 会改代码就自己做喽，这个问题是 kernel 分区过小所致的
> > dw33d 我的在学校，所以我只能下个月才能修下这个
> 
> 大佬，我只是小白类，不会代码，所以只能反馈，希望能得大佬们的帮助

真的，说换机器不是开玩笑的。这机器内核分区大小小的令人发指，即使这次AG大佬修了，说不定哪天更新内核大小又炸了，你在Issues搜下DW33D相关的问题，更新记录再搜下，这货都消减内核至少3次了。

给你指条路，还原官方Uboot，用Openwrt 主线代码得Ath79，那个将DW33D系统分区转移到了NAND Flash里面，内核大小也提高到4M

wifiway520, 2020-08-10:
> > > > 我是来反馈问题和寻找办法的！你可以洗洗了
> > > 
> > > 
> > > 会改代码就自己做喽，这个问题是 kernel 分区过小所致的
> > > dw33d 我的在学校，所以我只能下个月才能修下这个
> > 
> > 
> > 大佬，我只是小白类，不会代码，所以只能反馈，希望能得大佬们的帮助
> 
> 真的，说换机器不是开玩笑的。这机器内核分区大小小的令人发指，即使这次AG大佬修了，说不定哪天更新内核大小又炸了，你在Issues搜下DW33D相关的问题，更新记录再搜下，这货都消减内核至少3次了。
> 
> 给你指条路，还原官方Uboot，用Openwrt 主线代码得Ath79，那个将DW33D系统分区转移到了NAND Flash里面，内核大小也提高到4M
> 
> 我自己也有，内核砍了又砍，砍到1.7M，还是炸，索性出门左转扔垃圾桶了。

其实当初选这货也是图便宜，然后openwrt支持的很好，陆续的置办了4个，像你说的这些太专业，我也知道不容易，所以来这求助，能解决更好，4个直接左转有点浪费。。。再次感谢

AmadeusGhost, 2020-09-03:
建议直接刷潘多拉吧，op 很难再支持原来的启动方式了

tournet, 2021-01-16:
手上有几个dw33d，在用主线代码的ath79，将DW33D系统分区转移到了NAND Flash里面，固件可以随时用最新的，包括lean的也可以。Uboot是官方的还是Breed的都可以，breed的可以添加环境变量。

AmadeusGhost, 2021-02-27:
有大佬能测试下这个 commit 能正常启动嘛，breed 默认配置选择 factory.bin 刷入：https://github.com/AmadeusGhost/lede/commit/3ac4fa6762f0707d8347ace6794c934f490161cd

closed 2021-06-20 by coolsnowwolf