coolsnowwolf/lede #5300: 编译正常但无线无法使用

xxp2277, 2020-08-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
在https://github.com/coolsnowwolf/lede/pull/5295合并后拉取代码更新，正常编译结束。首次保留配置刷机后无线无法启动（登录luci界面发现无线所有选项不可用，删除/etc/config/wireless后luci界面的无线菜单不见）；不保留配置刷机启动后也看不见无线菜单（断电重启路由器后依旧无法看见）。
Q：是否搜索了issue (使用 "x" 选择)
* [√  ] 类似的issue
#5293 
## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：ar71xx平台，WNDR3800，2020-08-09代码

### (3) 详细日志

A：正常编译日志见https://github.com/xxp2277/WNDR3800/runs/963335020?check_suite_focus=true

panther706, 2020-08-09:
同样问题

AmadeusGhost, 2020-08-10:
日志指的是系统启动后的内核日志😂

panther706, 2020-08-10:
> 日志指的是系统启动后的内核日志😂

[    0.000000] Linux version 4.14.187 (pwh@pwh-machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2843-85808a075)) #0 Sun Aug 9 07:47:50 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9344 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 805adcc0, node_mem_map 81000020
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a4 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line:  board=WNDR4300 console=ttyS0,115200 mtdparts=ar934x-nfc:256k(u-boot)ro,256k(u-boot-env)ro,256k(caldata)ro,512k(pot),2048k(language),512k(config),3072k(traffic_meter),2048k(kernel),121856k(ubi),123904k@0x6c0000(firmware),256k(caldata_backup),-(reserved) rootfstype=squashfs noinitrd
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 123528K/131072K available (4309K kernel code, 182K rwdata, 944K rodata, 188K init, 209K bss, 7544K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] Clocks: CPU:560.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6825930166 ns
[    0.000010] sched_clock: 32 bits at 280MHz, resolution 3ns, wraps every 7669584382ns
[    0.008312] Calibrating delay loop... 278.93 BogoMIPS (lpj=1394688)
[    0.084943] pid_max: default: 32768 minimum: 301
[    0.090139] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.097207] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.107753] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.118271] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.125391] NET: Registered protocol family 16
[    0.131527] MIPS: machine is NETGEAR WNDR4300
[    0.136378] ar71xx: using random MAC address for eth0
[    0.394160] clocksource: Switched to clocksource MIPS
[    0.400803] NET: Registered protocol family 2
[    0.406311] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.413739] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.420564] TCP: Hash tables configured (established 1024 bind 1024)
[    0.427487] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.433720] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.440721] NET: Registered protocol family 1
[    0.445433] PCI: CLS 0 bytes, default 32
[    0.450820] Crashlog allocated RAM at address 0x3f00000
[    0.457982] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.471176] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.477436] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.502191] io scheduler noop registered
[    0.506422] io scheduler deadline registered (default)
[    0.512253] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.519454] console [ttyS0] disabled
[    0.543338] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.552538] console [ttyS0] enabled
[    0.559966] bootconsole [early0] disabled
[    0.572936] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xf1
[    0.579439] nand: Micron NAND 128MiB 3,3V 8-bit
[    0.584042] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[... 144 lines cut ...]
[   19.486027] PPP MPPE Compression module registered
[   19.505940] NET: Registered protocol family 24
[   19.637911] mac80211: Unknown symbol netif_receive_skb_list (err 0)
[   19.674811] kmodloader: 2 modules could not be probed
[   19.680061] kmodloader: dependency not loaded mac80211
[   19.694366] kmodloader: - ath9k - 1
[   19.698103] kmodloader: - mac80211 - 0
[   69.523016] print_req_error: I/O error, dev mtdblock0, sector 48
[   69.558254] print_req_error: I/O error, dev mtdblock0, sector 88
[   69.711163] print_req_error: I/O error, dev mtdblock4, sector 88
[   76.190052] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   76.216335] device eth0 entered promiscuous mode
[... cut ...]
[   76.330139] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
[   77.216045] eth0: link up (1000Mbps/Full duplex)
[   77.220810] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   77.294329] br-lan: port 1(eth0.1) entered blocking state
[   77.299825] br-lan: port 1(eth0.1) entered forwarding state
[   77.305886] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
[   77.464748] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   78.835606] fast-classifier (PBR safe v2.1.4a): starting up
[   78.841467] fast-classifier: registered

rockey2020, 2020-08-10:
我也是 ar750s 型号

qq635938204, 2020-08-10:
树莓派3、4也存在同样问题

AmadeusGhost, 2020-08-10:
> 树莓派3、4也存在同样问题

你确定？？？这是 4.14，不是 5.4
> mac80211: Unknown symbol netif_receive_skb_list (err 0)

I'll backport this.

biozzl, 2020-08-11:
我也是无线不可用呢， MW4530R, 老版本编译就没问题

PR #5303 (2020-08-11): mac80211: support kernel 4.14 and update to 5.8
fix #5300
closed 2020-08-11 by coolsnowwolf