coolsnowwolf/lede #5302: LXC下运行端口转发问题

imEgo, 2020-08-09:
PVE直接用LXC来跑常用功能正常，但是内网通过域名加端口，走端口转发无法访问内网其他设备（openwrt本身可以），而外网通过此方式访问则全部正常。
于是开始排查问题，在目标设备通过tcpdump抓不到包；然后在openwrt上使用tcpdump抓包，此时内网通过域名加端口走端口转发恢复正常，关闭tcpdump后则问题依旧。后续使用QEMU跑的话相同配置功能正常，奈何LXC相比QEMU性能好太多，500M带宽跑满，CPU也只有不到10%的占用。
顾还是想问下此问题有无思路解决？或是如何进一步排查原因？

ElectronicElephant, 2020-08-10:
这个功能叫做 NAT Loopback，借用 OpenWRT 论坛上的话
https://forum.openwrt.org/t/help-with-hairpinning-nat-loopback-with-openwrt-18-06/62893/2

> NAT Loopback is not a good solution.
> Just create a domain entry in dhcp configuration (or Network Hostnames in Luci) with the name that you use and the internal IP of the server.

直接在 `网络` - `DHCP/DNS` - `自定义挟持域名` 中填入你的内网地址即可。

imEgo, 2020-08-10:
> 这个功能叫做 NAT Loopback，借用 OpenWRT 论坛上的话
> https://forum.openwrt.org/t/help-with-hairpinning-nat-loopback-with-openwrt-18-06/62893/2
>
>
> 直接在 `网络` - `DHCP/DNS` - `自定义挟持域名` 中填入你的内网地址即可。

感谢解答，自定义挟持域名将域名配置成了192.168.2.1（OpenWrt的内网地址），同时端口转发添加了一组Lan到Lan的规则，问题依旧，请问是端口转发配置的问题么？

ElectronicElephant, 2020-08-11:
> 感谢解答，自定义挟持域名将域名配置成了192.168.2.1（OpenWrt的内网地址），同时端口转发添加了一组Lan到Lan的规则，问题依旧，请问是端口转发配置的问题么？

额，不是这样填的。比如你的NAS的内网地址是 192.168.2.5，你的域名是 nas.yumk.org，那么就把 nas.yumk.org 挟持到 192.168.2.5。这其实就是相当于一个自定义DNS ，这么说你明白了吧

imEgo, 2020-08-11:
> 额，不是这样填的。比如你的NAS的内网地址是 192.168.2.5，你的域名是 nas.yumk.org，那么就把 nas.yumk.org 挟持到 192.168.2.5。这其实就是相当于一个自定义DNS ，这么说你明白了吧

懂了，但是不能满足我的使用场景，我现在是一个域名通过不同端口转发到不同设备上的不同服务，不过仍然感谢，目前只能后台起一个tcpdump一直抓包暂时解决问题了

closed 2021-06-20 by coolsnowwolf