coolsnowwolf/lede #5304: j4105 cpu，长期高频，低负载依然处于睿频状态

rfigo, 2020-08-10:
(1) 具体问题

A：试过4.19和5.4内核，开发版 稳定版都进行了测试，发现cpu会长时间处于睿频状态，cpu负载并不高，但是cpu长时间高频（1800-2300mhz）

(2) 路由器型号和固件版本

A：intel j4105 cpu，4g ram
cpu频率800-1500mhz，可睿频2500mhz
lienol snapshot源码固件正常，9号之后的lienol snapshot源码也出现类似问题

hcym, 2020-08-10:
负载不高温度不高就行了，我是从来没指望过什么睿智的频率

rfigo, 2020-08-10:
> 负载不高温度不高就行了，我是从来没指望过什么睿智的频率

有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）

dolphinpaopao, 2020-08-10:
> > 负载不高温度不高就行了，我是从来没指望过什么睿智的频率
> 
> 有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）

现在同步的是官方，有问题要去官方反馈了，况且只是凭借一个现象描述如何定位解决问题？新版有问题就回退到没有问题的版本。等待官方修复。

rfigo, 2020-08-10:
> > > 负载不高温度不高就行了，我是从来没指望过什么睿智的频率
> > 
> > 
> > 有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）
> 
> 现在同步的是官方，有问题要去官方反馈了，况且只是凭借一个现象描述如何定位解决问题？新版有问题就回退到没有问题的版本。等待官方修复。

同步官方源码的新版问题反而没那么严重，起码cpu会调频，刷过大雕一个稳定版源码的固件（4.19+luci 17.01），反而cpu卡高频不会升降了。
回退可以用的版本肯定是会刷的，只是反馈一下，觉得没用就关了吧

dongjuanyong, 2020-08-11:
> > > 负载不高温度不高就行了，我是从来没指望过什么睿智的频率
> > 
> > 
> > 有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）
> 
> 现在同步的是官方，有问题要去官方反馈了，况且只是凭借一个现象描述如何定位解决问题？新版有问题就回退到没有问题的版本。等待官方修复。

https://github.com/coolsnowwolf/lede/blob/master/target/linux/x86/config-5.4
https://github.com/openwrt/openwrt/blob/master/target/linux/x86/config-5.4
你去比较一下lede和openwrt官方的kernel config，问题极有可能出在同步以外的个别优化上面

dolphinpaopao, 2020-08-11:
> > > > 负载不高温度不高就行了，我是从来没指望过什么睿智的频率
> > > 
> > > 
> > > 有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）
> > 
> > 
> > 现在同步的是官方，有问题要去官方反馈了，况且只是凭借一个现象描述如何定位解决问题？新版有问题就回退到没有问题的版本。等待官方修复。
> 
> 同步官方源码的新版问题反而没那么严重，起码cpu会调频，刷过大雕一个稳定版源码的固件（4.19+luci 17.01），反而cpu卡高频不会升降了。
> 回退可以用的版本肯定是会刷的，只是反馈一下，觉得没用就关了吧

反馈bug需要有详细的日志，使用环境，后台运行程序等等，问题能复现最好，以便于定位问题。只有一句话的现象描述，神仙才知道问题在哪里。现在我还用着七月初未同步到官方的版本，upnp也是作者修改过的最新2.1，迅雷比特彗星手机等软件和设备，端口映射工作都很OK。没有出现其他人一直反馈的upnp2.1新版异常，所以不同环境差别太大。

aglent, 2020-08-11:
我也有这个问题 J4105 默认1.5G频率 运行时基本在1.8-2.2G之间
另一个N3540的就正常

rfigo, 2020-08-11:
> 反馈bug需要有详细的日志，使用环境，后台运行程序等等，问题能复现最好，以便于定位问题。只有一句话的现象描述，神仙才知道问题在哪里。现在我还用着七月初未同步到官方的版本，upnp也是作者修改过的最新2.1，迅雷比特彗星手机等软件和设备，端口映射工作都很OK。没有出现其他人一直反馈的upnp2.1新版异常，所以不同环境差别太大。

看一下楼下 同款cpu一样问题 cpu调度不是网络参数设置 没有那么大的环境干扰要素的

Promix953, 2020-08-11:
> 看一下楼下 同款cpu一样问题 cpu调度不是网络参数设置 没有那么大的环境干扰要素的

编译所用config呢?
内核和系统日志呢?
软路由硬件配置呢?
网络结构呢?
以上四点给一个也行啊，难道要开发者跑你家里给你现场解决么?
你看病难道就打电话给医生，一句我头疼就叫医生给你治么?

dongjuanyong, 2020-08-12:
行了，我给楼主一个思路 41965f2d26de01be34f83c7ba1fd3ddc4d3152f6 这个commit删掉了kernel config里面的CONFIG_INTEL_IDLE=y ，可能是导致问题的原因，把这个commit revert了你再试试

上面的也别说风凉话，一个cpu频率调度的问题你从内核和系统日志能看出来啥？
你去医院看病大夫也没让你拿着DNA检测报告才能进门是不

rfigo, 2020-08-12:
> > > > > > > 负载不高温度不高就行了，我是从来没指望过什么睿智的频率
> > > > > > 
> > > > > > 
> > > > > > 有问题就是有问题，这种事情没有模棱两可这回事的，就按你说的负载和温度这两点来说，有bug的负载和温度都会高于没有bug的版本（比如lienol snapshot 8月8日之前的）
> > > > > 
> > > > > 
> > > > > 现在同步的是官方，有问题要去官方反馈了，况且只是凭借一个现象描述如何定位解决问题？新版有问题就回退到没有问题的版本。等待官方修复。
> > > > 
> > > > 
> > > > 同步官方源码的新版问题反而没那么严重，起码cpu会调频，刷过大雕一个稳定版源码的固件（4.19+luci 17.01），反而cpu卡高频不会升降了。
> > > > 回退可以用的版本肯定是会刷的，只是反馈一下，觉得没用就关了吧
> > > 
> > > 
> > > 反馈bug需要有详细的日志，使用环境，后台运行程序等等，问题能复现最好，以便于定位问题。只有一句话的现象描述，神仙才知道问题在哪里。现在我还用着七月初未同步到官方的版本，upnp也是作者修改过的最新2.1，迅雷比特彗星手机等软件和设备，端口映射工作都很OK。没有出现其他人一直反馈的upnp2.1新版异常，所以不同环境差别太大。
> > 
> > 
> > 看一下楼下 同款cpu一样问题 cpu调度不是网络参数设置 没有那么大的环境干扰要素的
> 
> 编译所用config呢?
> 内核和系统日志呢?
> 软路由具体硬件配置呢?
> 网络结构呢?
> 给个日志也比一句J4105不降频强啊，难道要开发者跑你家里给你现场解决么?
> 你看病难道就打电话给医生，一句我头疼就叫医生给你治么?

幸亏问题不是反馈给你的，不然你不得上天？
cpu调频日志能看出来？如果报err了我能不贴吗，真是神奇

rfigo, 2020-08-12:
> 行了，我给楼主一个思路 [41965f2](https://github.com/coolsnowwolf/lede/commit/41965f2d26de01be34f83c7ba1fd3ddc4d3152f6) 这个commit删掉了kernel config里面的CONFIG_INTEL_IDLE=y ，可能是导致问题的原因，把这个commit revert了你再试试
> 
> 上面的也别说风凉话，一个cpu频率调度的问题你从内核和系统日志能看出来啥？
> 你去医院看病大夫也没让你拿着DNA检测报告才能进门是不

非常感谢！！！

dolphinpaopao, 2020-08-12:
我是物理机安装i3 4150T，刚刚编译测试了最新版本，频率调节正常可以下探到0.9GHz，怀疑应该是某一个功能在后台占用导致的问题。开启或者关闭intel idle没有影响。

Promix953, 2020-08-12:
上面两位，反对一句话issue、无基本信息issue就是上天?就是要DNA测序?
知道为啥家用电器说明书里故障排除第一条是检查电源线这种看起来很傻的提示么?

dongjuanyong, 2020-08-12:
> 上面两位，反对一句话issue、无基本信息issue就是上天?就是要DNA测序?
> 知道为啥家用电器说明书里故障排除第一条是检查电源线这种看起来很傻的提示么?

大家都是幽默两句，不识逗就没意思了不是
我承认issue区很多小白提的问题很没有营养也缺乏线索，但从lz的提问当中明显能感受到一定水平，至少明确意识到cpu频率调节问题靠简单的log定位基本实现不了。
同步官方也不是100%，你要是认为官方只会做减法只能靠L大做加法，那这bug fix的活就没指望了

关于 INTEL_IDLE
```
Enable intel_idle, a cpuidle driver that includes knowledge of native Intel hardware idle features.
The acpi_idle driver can be configured at the same time, in order to handle processors intel_idle does not support.
```
ATOM系很多feature的实现和主流桌面平台不太一样，可能深度的cpu频率控制是在ACPI之外实现的
我用的是n3710，kernel config是CONFIG_INTEL_IDLE=y，于是在/sys/devices/system/cpu/cpuidle/current_driver中明确显示intel_idle，cpuinfo_min_freq能达到480000

rfigo, 2020-08-12:
> 大家都是幽默两句，不识逗就没意思了不是
> 我承认issue区很多小白提的问题很没有营养也缺乏线索，但从lz的提问当中明显能感受到一定水平，至少明确意识到cpu频率调节问题靠简单的log定位基本实现不了。
> 同步官方也不是100%，你要是认为官方只会做减法只能靠L大做加法，那这bug fix的活就没指望了
>
> 关于 INTEL_IDLE
>
> ```
> Enable intel_idle, a [...]

水平完全没有，软路由完全是小白，不过之前喜欢刷手机内核，对高通soc的cpu调度有一点点了解，所以从系统日志和内核日志看不出什么问题之后，推测和手机内核差不多，可能是cpu idle失效或者有异常占用，就反馈了一下，没想到遇到某牛批的不得了的人物上来就给你开方子让体检自测然后拿着报告再来，可能这货对cpu调频策略根本一无所知才把这事当检查电源线吧，幸亏有您这种高知又耐心的大佬指点，不多说了，感谢

helianling, 2020-08-13:
或许可以先编译一个最精简固件试试，这样可以检测到底是内核导致的，还是由于某些软件引起的，初步排除吧。

smalldickman, 2020-08-13:
会不定时重启吗，我的淘宝买的j4105也是一直高频，换了一个新的频率倒是可以降了

aglent, 2020-08-13:
问下 ssh执行cpuinfo 为什么显示的频率比openwrt显示出来的要低很多？

rfigo, 2020-08-13:
> 会不定时重启吗，我的淘宝买的j4105也是一直高频，换了一个新的频率倒是可以降了

不会定时重启，有些版本固件没问题，说明硬件ok

smalldickman, 2020-08-13:
> > 会不定时重启吗，我的淘宝买的j4105也是一直高频，换了一个新的频率倒是可以降了
> 
> 不会定时重启，有些版本固件没问题，说明硬件ok

我的老是随机重启换各种固件都会，影响使用就换了。。

rfigo, 2020-08-13:
> > > 会不定时重启吗，我的淘宝买的j4105也是一直高频，换了一个新的频率倒是可以降了
> > 
> > 
> > 不会定时重启，有些版本固件没问题，说明硬件ok
> 
> 我的老是随机重启换各种固件都会，影响使用就换了。。

那应该是硬件问题 和此issues无关了

rfigo, 2020-08-13:
最新进展，似乎是luci显示的bug，用cpuinfo命令查看，就是正常的……如图

[image]

rfigo, 2020-08-13:
> 大家都是幽默两句，不识逗就没意思了不是
> 我承认issue区很多小白提的问题很没有营养也缺乏线索，但从lz的提问当中明显能感受到一定水平，至少明确意识到cpu频率调节问题靠简单的log定位基本实现不了。
> 同步官方也不是100%，你要是认为官方只会做减法只能靠L大做加法，那这bug fix的活就没指望了
>
> 关于 INTEL_IDLE
>
> ```
> Enable intel_idle, a [...]

最新进展，似乎是luci显示的bug，用cpuinfo命令查看，就是正常的……如图
[image]

rfigo, 2020-08-13:
> 问下 ssh执行cpuinfo 为什么显示的频率比openwrt显示出来的要低很多？

确实如此，跑了下负载，cpuinfo显示是准确的，会升降频，luci界面显示的不准，应该是bug了

Pastechn, 2020-09-12:
我也遇到了这个问题 luci 上的 cpu 频率居高不下 后来发现是只要打开 luci 后台界面 无论是页面上还是 cpuinfo 显示的频率都很高 并且负载直线上升 一旦关掉 luci 后台界面之后 cpuinfo 的频率马上掉到 800MHz 负载也下来了

closed 2021-06-20 by coolsnowwolf

kxdn3, 2023-12-14:
j4125 6.1内核同故障