coolsnowwolf/lede #5305: Samba4是不是不支持SMBv1了？

Gokou-Ruri, 2020-08-10:
1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无论怎么改参数也不能用 SMBv1 访问共享，试过 `server min protocol = NT1` (参数的值也试过 SMB1 和 SMBv1 ) ，但没有效果，依旧无法连接 SMBv1 共享（只支持 SMBv1 的手机 APP ），除此之外也试过直接指定 `protocol = NT1` ，也无效。请教要如何让 Samba4 启用 SMBv1 ？还是说最新版的 Samba4 已经不支持 SMBv1 了？

### (2) 路由器型号和固件版本

A：Newifi-D2 OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13)

### (3) 详细日志

A：NONE

coolsnowwolf, 2020-08-14:
v1 太老了，默认没开

Gokou-Ruri, 2020-08-15:
> v1 太老了，默认没开

大佬可以讲一下怎么开启吗？我这边使用上面的参数配置无法开启。

Rongronggg9, 2020-08-23:
```
[global]
	netbios name = |NAME| 
	interfaces = |INTERFACES|
	server string = |DESCRIPTION|
	unix charset = |CHARSET|
	workgroup = |WORKGROUP|

	## This global parameter allows the Samba admin to limit what interfaces on a machine will serve SMB requests.
	bind interfaces only = yes

	## time for inactive connections to-be closed in minutes
	deadtime = 15

	## disable core dumps
	enable core files = no

	## set security (auto, user, domain, ads)
	security = user

	## This parameter controls whether a remote client is allowed or required to use SMB encryption.
	## It has different effects depending on whether the connection uses SMB1 or SMB2 and newer:
    ## If the connection uses SMB1, then this option controls the use of a Samba-specific extension to the SMB protocol introduced in Samba 3.2 that makes use of the Unix extensions.
	## If the connection uses SMB2 or newer, then this option controls the use of the SMB-level encryption that is supported in SMB version 3.0 and above and available in Windows 8 and newer. 
	## (default/auto,desired,required,off)
	#smb encrypt = auto

	server min protocol = NT1

	## set invalid users
	invalid users = root

	## map unknow users to guest
	map to guest = Bad User

	## allow client access to accounts that have null passwords. 
	null passwords = yes

	## The old plaintext passdb backend. Some Samba features will not work if this passdb backend is used. (NOTE: enabled for size reasons)
	## (tdbsam,smbpasswd,ldapsam)
	passdb backend = smbpasswd

	## Set location of smbpasswd ('smbd -b' will show default compiled location)
	#smb passwd file = /etc/samba/smbpasswd 

	## LAN/WAN options (IPTOS_LOWDELAY TCP_NODELAY) WAN (IPTOS_THROUGHPUT)
	socket options = IPTOS_LOWDELAY TCP_NODELAY

	## If this integer parameter is set to a non-zero value, Samba will read from files asynchronously when the request size is bigger than this value.
	## Note that it happens only for non-chained and non-chaining reads and when not using write cache.
	## The only reasonable values for this parameter are 0 (no async I/O) and 1 (always do async I/O).
	## (1/0)
	#aio read size = 0
	#aio write size = 0

	## If Samba has been built with asynchronous I/O support, Samba will not wait until write requests are finished before returning the result to the client for files listed in this parameter.
	## Instead, Samba will immediately return that the write request has been finished successfully, no matter if the operation will succeed or not.
	## This might speed up clients without aio support, but is really dangerous, because data could be lost and files could be damaged. 
	#aio write behind = /*.tmp/

	## lower CPU useage if supported and aio is disabled (aio read size = 0 ; aio write size = 0)
	## currently broken (NOTE: see https://bugzilla.samba.org/show_bug.cgi?id=14095 )
	## (no, yes)
	#use sendfile = yes

	## samba will behave as previous versions of Samba would and will fail the lock request immediately if the lock range cannot be obtained.
	#blocking locks = No

	## disable loading of all printcap printers by default (iprint, cups, lpstat)
	load printers = No
	printcap name = /dev/null

	## Enabling this parameter will disable Samba's support for the SPOOLSS set of MS-RPC's.
	disable spoolss = yes

	## This parameters controls how printer status information is interpreted on your system.
	## (BSD, AIX, LPRNG, PLP, SYSV, HPUX, QNX, SOFTQ)
	printing = bsd

	## Disable that nmbd is acting as a WINS server for unknow netbios names
	#dns proxy = No

	## win/unix user mapping backend
	#idmap config * : backend = tdb

	## Allows the server name that is advertised through MDNS to be set to the hostname rather than the Samba NETBIOS name.
	## This allows an administrator to make Samba registered MDNS records match the case of the hostname rather than being in all capitals.
	## (netbios, mdns)
	mdns name = mdns

	## Clients that only support netbios won't be able to see your samba server when netbios support is disabled.
	#disable netbios = Yes

	## Setting this value to no will cause nmbd never to become a local master browser.
	#local master = no

	## (auto, yes) If this is set to yes, on startup, nmbd will force an election, and it will have a slight advantage in winning the election. It is recommended that this parameter is used in conjunction with domain master = yes, so that nmbd can guarantee becoming a domain master. 
	#preferred master = yes

	## (445 139) Specifies which ports the server should listen on for SMB traffic.
	## 139 is netbios/nmbd
	#smb ports = 445 139

	## This is a list of files and directories that are neither visible nor accessible.
	## Each entry in the list must be separated by a '/', which allows spaces to be included in the entry. '*' and '?' can be used to specify multiple files or directories as in DOS wildcards.
	veto files = /Thumbs.db/.DS_Store/._.DS_Store/.apdisk/

	## If a directory that is to be deleted contains nothing but veto files this deletion will fail unless you also set the delete veto files parameter to yes.
	delete veto files = yes

################ Filesystem and creation rules ################
	## reported filesystem type (NTFS,Samba,FAT)
	#fstype = FAT

	## Allows a user who has write access to the file (by whatever means, including an ACL permission) to modify the permissions (including ACL) on it.
	#dos filemode = Yes

	## file/dir creating rules
	create mask = 0777
	directory mask = 0777
	#force group = root
	#force user = root
	#inherit owner = windows and unix
################################################################

```
供参考。

klever1988, 2020-08-24:
有些蓝光播放机上的smb协议还是v1版的

Gokou-Ruri, 2020-08-25:
> ```
> 	security = user
> 	server min protocol = NT1
> ```
> 
> 供参考。

谢谢大佬，我重新编译了固件，现在可以使用SMBv1了。

closed 2020-08-25 by Gokou-Ruri