coolsnowwolf/lede #5306: 请教一下编译树莓派64位和R2s固件cpu架构为问号的问题

trombonist852, 2020-08-11:
如题请教一下各位大神，树莓派32位就可以正常显示，64位就不行，r2s也是一样，auto_core.diff的补丁也打过了，menuconfig里面autocore也选择了，可就是不显示，请大神们帮忙分析一下是什么原因？是不是还要做一些地方的改动？万分感激！！

AmadeusGhost, 2020-08-11:
Just run `./scripts/feeds update -a`, I fixed this in luci.

klever1988, 2020-08-11:
系统->软件包
里面看看，确认autocore是否安装了

trombonist852, 2020-08-11:
> 已经安装了…

klever1988, 2020-08-11:
那你看一下我编译的固件和脚本吧，我是正常的

https://github.com/klever1988/lede/actions/runs/203788560

trombonist852, 2020-08-11:
> 那你看一下我编译的固件和脚本吧，我是正常的
> 
> https://github.com/klever1988/lede/actions/runs/203788560

好的！感谢大神！

trombonist852, 2020-08-11:
> 那你看一下我编译的固件和脚本吧，我是正常的
>
>
>
> https://github.com/klever1988/lede/actions/runs/203788560
>
>
>
>
>
>
> 对了，麻烦问一下，您这个autocore是打过补丁的不？—

klever1988, 2020-08-11:
> […](#)

打补丁的意义在于让autocore在arm平台成为可选编译安装，不打补丁的话只能在编译x86时选择autocore
显示cpu频率和温度需要autocore

trombonist852, 2020-08-11:
> > […](#)
> 
> 打补丁的意义在于让autocore在arm平台成为可选编译安装，不打补丁的话只能在编译x86时选择autocore
> 显示cpu频率和温度需要autocore
我看了一下您的config里面也只是选了autocore，其他的也没选，是不是要修改autocore里面的cpuinfo这个文件？

klever1988, 2020-08-11:
> > > […](#)
> > 
> > 
> > 打补丁的意义在于让autocore在arm平台成为可选编译安装，不打补丁的话只能在编译x86时选择autocore
> > 显示cpu频率和温度需要autocore
> > 我看了一下您的config里面也只是选了autocore，其他的也没选，是不是要修改autocore里面的cpuinfo这个文件？

只要打了补丁选了autocore就可以显示频率和温度 除非你打的补丁不对。
你发一下你的补丁

trombonist852, 2020-08-11:
> > > > […](#)
> > > 
> > > 
> > > 打补丁的意义在于让autocore在arm平台成为可选编译安装，不打补丁的话只能在编译x86时选择autocore
> > > 显示cpu频率和温度需要autocore
> > > 我看了一下您的config里面也只是选了autocore，其他的也没选，是不是要修改autocore里面的cpuinfo这个文件？
> 
> 只要打了补丁选了autocore就可以显示频率和温度 除非你打的补丁不对。
> 你发一下你的补丁
https://github.com/klever1988/lede/blame/master/enable_autocore.diff
就是您提交的这个补丁，不过我用的lienol大神整合过的lean大神的源码，我对比过lean大神官方的源码和lienol大神整合过的源码，两个autocore里面的cpuinfo内容不一样，我试过直接把lean大神源码里的
autocore替换进去，但是编译出来之后就直接luci界面跳错误....我能否把现在我用的没有打过补丁的autocore文件打个包传上来您方便的时候帮我看看么？这个问题困扰我好久了，一直没找到方法解决，网上几乎都搜不到....
[autocore.zip](https://github.com/coolsnowwolf/lede/files/5058230/autocore.zip)

klever1988, 2020-08-11:
[enable_autocore-Lienols-Ver.diff.txt](https://github.com/coolsnowwolf/lede/files/5058345/enable_autocore-Lienols-Ver.diff.txt)

做了个Lienol版本的补丁 试试看

trombonist852, 2020-08-11:
> [enable_autocore-Lienols-Ver.diff.txt](https://github.com/coolsnowwolf/lede/files/5058345/enable_autocore-Lienols-Ver.diff.txt)
> 
> 做了个Lienol版本的补丁 试试看
大神 太感谢了！！！！！我这就去试试看！

trombonist852, 2020-08-11:
> [enable_autocore-Lienols-Ver.diff.txt](https://github.com/coolsnowwolf/lede/files/5058345/enable_autocore-Lienols-Ver.diff.txt)
> 
> 做了个Lienol版本的补丁 试试看

对了大神，再麻烦问一下，这个补丁同样适用于树莓派的64位版本么？？

trombonist852, 2020-08-11:
> > [enable_autocore-Lienols-Ver.diff.txt](https://github.com/coolsnowwolf/lede/files/5058345/enable_autocore-Lienols-Ver.diff.txt)
> > 做了个Lienol版本的补丁 试试看
> 
> 对了大神，再麻烦问一下，这个补丁同样适用于树莓派的64位版本么？？
大神，刷好了
现在可以显示频率了  但是没有温度显示，这个如何解决？？

trombonist852, 2020-08-11:
> [enable_autocore-Lienols-Ver.diff.txt](https://github.com/coolsnowwolf/lede/files/5058345/enable_autocore-Lienols-Ver.diff.txt)
> 
> 做了个Lienol版本的补丁 试试看

大神，刷好了
现在可以显示频率了 但是没有温度显示，这个如何解决？？

wushenghn, 2020-08-12:
温度可以自己加
/usr/lib/lua/luci/view/admin_status/index.htm  704行
<tr><td width="33%"><%:CPU温度%></td><td id=""><%=luci.sys.exec("echo $((`cat /sys/class/thermal/thermal_zone0/temp` / 1000))'℃'") or "?" %></td></tr>``

trombonist852, 2020-08-12:
> <%:CPU温度%><%=luci.sys.exec("echo $((`cat
> 感谢大神！我去试一下！—

trombonist852, 2020-08-12:
> 温度可以自己加
> /usr/lib/lua/luci/view/admin_status/index.htm 704行
> 
> <%:CPU温度%><%=luci.sys.exec("echo $((`cat /sys/class/thermal/thermal_zone0/temp` / 1000))'℃'") or "?" %>``

大神，出问题了，按照您说的我打开这个文件为啥我这儿
[index.zip](https://github.com/coolsnowwolf/lede/files/5060440/index.zip)

最多只有78行就到底了........

wushenghn, 2020-08-12:
路径不对
root@OpenWrt:/usr/lib/lua/luci/view/admin_status# ls
bandwidth.htm    dmesg.htm        index.htm        load.htm         syslog.htm
connections.htm  index            iptables.htm     routes.htm       wireless.htm
root@OpenWrt:/usr/lib/lua/luci/view/admin_status# head -728 index.htm | tail -20
<h2 name="content"><%:Status%></h2>

<fieldset class="cbi-section">
	<legend><%:System%></legend>

	<table width="100%" cellspacing="10">
		<tr><td width="33%"><%:Hostname%></td><td><%=luci.sys.hostname() or "?"%></td></tr>
		<tr><td width="33%"><%:Model%></td><td><%=pcdata(boardinfo.model or "?")%> <%=luci.sys.exec("cat /etc/bench.log") or " "%></td></tr>
		<tr><td width="33%"><%:CPU Info%></td><td id="cpuinfo">-</td></tr>
		<tr><td width="33%"><%:Firmware Version%></td><td>
			<%=pcdata(ver.distname)%> <%=pcdata(ver.distversion)%> /
			<%=pcdata(ver.luciname)%> (<%=pcdata(ver.luciversion)%>)
		</td></tr>
		<tr><td width="33%"><%:Kernel Version%></td><td><%=unameinfo.release or "?"%></td></tr>
		<tr><td width="33%"><%:Local Time%></td><td id="localtime">-</td></tr>
		<tr><td width="33%"><%:Uptime%></td><td id="uptime">-</td></tr>
		<tr><td width="33%"><%:Load Average%></td><td id="loadavg">-</td></tr>
		<tr><td width="33%"><%:CPU usage (%)%></td><td id="cpuusage">-</td></tr>
这里加一行
	</table>
</fieldset>
root@OpenWrt:/usr/lib/lua/luci/view/admin_status#

trombonist852, 2020-08-12:
> 路径不对
> root@OpenWrt:/usr/lib/lua/luci/view/admin_status# ls
> bandwidth.htm dmesg.htm index.htm load.htm syslog.htm
> connections.htm index iptables.htm routes.htm wireless.htm
> 
> **_index.htm_** 700多行
> `<tr><td width="33%"><%:CPU usage (%)%></td><td id="cpuusage">-</td></tr> [image] `
> 这里加一行

好的，感谢大神！我再去找一下！那如果我想集成到源码里面的话，这个文件是在源码里的哪个目录下？我集成进去，省的每次刷机都要改了....

wushenghn, 2020-08-12:
package/lean/autocore/files/index.htm

trombonist852, 2020-08-12:
> 路径不对
> ‘
> root@OpenWrt:/usr/lib/lua/luci/view/admin_status# ls
> bandwidth.htm dmesg.htm index.htm load.htm syslog.htm
> connections.htm index iptables.htm routes.htm wireless.htm
> ’
> 
> **_index.htm_** 700多行
> `<%:CPU usage (%)%>-’
> 这里加一行

大神，我刚才看了一下:/usr/lib/lua/luci/view/admin_status这个文件夹里面没有您上面ls出来的这么多文件 只有一个index.htm....

trombonist852, 2020-08-12:
> package/lean/autocore/files/index.htm

好的！感谢大神！！！我去添加去！

trombonist852, 2020-08-12:
> package/lean/autocore/files/index.htm

我知道问题了，我这里package/lean/autocore/files/里面没有index.htm这个文件，我用的是lienol大神整合的源码，我去lean的源码里拖一份index放进去应该也可以的吧？？

wushenghn, 2020-08-12:
文件位置不一样自己搜索一下

klever1988, 2020-08-12:
autocore只是从系统提取温度保存到系统变量，而luci（Lean大版本）自带提取显示这个变量
Lienol版本需要自己改动feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm这个文件显示变量

trombonist852, 2020-08-12:
> autocore只是从系统提取温度保存到系统变量，而luci（Lean大版本）自带提取显示这个变量
> Lienol版本需要自己改动feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm这个文件显示变量

好的大神！！我去改一下！！太感谢了！！！

klever1988, 2020-08-12:
试一下在feeds install之后执行

sed -i '/Load Average/i\<div class="tr"><div class="td left" width="33%"><%:CPU Temperature%></div><div class="td left" id="_cputemp"><%=luci.sys.exec("cut -c1-2 /sys/class/thermal/thermal_zone0/temp")%></div></div>' feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm

trombonist852, 2020-08-12:
> autocore只是从系统提取温度保存到系统变量，而luci（Lean大版本）自带提取显示这个变量
> Lienol版本需要自己改动feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm这个文件显示变量
大神，您说的这个路径里面只有一个makefile文件 然后我打开看里面的内容，有一行是关于依赖LUCI_DEPENDS:=+luci-base +luci-mod-status +luci-mod-system +luci-mod-network
然后我看了一下luci-mod-status里面是有您说的这个文件的，我需要把您说的这个路径改成feeds/luci/modules/luci-mod-status/luasrc/view/admin_status/index.htm么？

trombonist852, 2020-08-12:
> 试一下在feeds install之后执行
> 
> sed -i '/Load Average/i<div class="tr">
> 
> <%:CPU Temperature%>
> <%=luci.sys.exec("cut -c1-2 /sys/class/thermal/thermal_zone0/temp")%>
> ' feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm

这个是luci-mod-admin-full/makefile的所有内容：
#
# Copyright (C) 2008-2014 The LuCI Team <luci@lists.subsignal.org>
#
# This is free software, licensed under the Apache License, Version 2.0 .
#

include $(TOPDIR)/rules.mk

LUCI_TITLE:=LuCI Administration - full-featured for full control
LUCI_DEPENDS:=+luci-base +luci-mod-status +luci-mod-system +luci-mod-network

PKG_LICENSE:=Apache-2.0

include ../../luci.mk

# call BuildPackage - OpenWrt buildroot signature

klever1988, 2020-08-12:
先feed install -a才有htm文件

trombonist852, 2020-08-12:
我执行./scripts/feeds install -a啦...但是还是没有.....

wangjun@Trb:~/op-dev-master$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed luci.
Installing all packages from feed diy1.
Installing all packages from feed freifunk.

> 先feed install -a才有htm文件

trombonist852, 2020-08-12:
> 先feed install -a才有htm文件

哦我知道为什么了 li大神的luci-19.07没有这个文件18.06的有，我去切换一下，感谢大神！

klever1988, 2020-08-12:
Lienol的openwrt默认分支是dev-19.07，默认的luci也是19.07，为什么你都选非主流的分支啊

https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default
...
src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07

trombonist852, 2020-08-12:
> Lienol的openwrt默认分支是dev-19.07，默认的luci也是19.07，为什么你都选非主流的分支啊
> 
> https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default
> ...
> src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07

主要是想学习学习，看看在编译的过程中会遇到啥问题，怎么解决，然后是想用用新内核，说白了就是找虐🤣🤣🤣

trombonist852, 2020-08-12:
> Lienol的openwrt默认分支是dev-19.07，默认的luci也是19.07，为什么你都选非主流的分支啊
> 
> https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default
> ...
> src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07

他的luci-19.07就没有您说的那个文件，切换到18就有了

klever1988, 2020-08-12:
> > Lienol的openwrt默认分支是dev-19.07，默认的luci也是19.07，为什么你都选非主流的分支啊
> > https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default
> > ...
> > src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07
> 
> 他的luci-19.07就没有您说的那个文件，切换到18就有了

有的，你需要执行这两个命令
./scripts/feeds update -a
./scripts/feeds install -a
之后才会有

trombonist852, 2020-08-12:
> 有的，你需要执行这两个命令
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 之后才会有

https://github.com/Lienol/openwrt-luci/tree/dev-19.07/modules/luci-mod-admin-full
您看看 真的没有.......

klever1988, 2020-08-12:
> > > > Lienol的openwrt默认分支是dev-19.07，默认的luci也是19.07，为什么你都选非主流的分支啊
> > > > https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default
> > > > ...
> > > > src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07
> > > 
> > > 
> > > 他的luci-19.07就没有您说的那个文件，切换到18就有了
> > 
> > 
> > 有的，你需要执行这两个命令
> > ./scripts/feeds update -a
> > ./scripts/feeds install -a
> > 之后才会有
> 
> https://github.com/Lienol/openwrt-luci/tree/dev-19.07/modules/luci-mod-admin-full
> 您看看 真的没有.......
> https://github.com/Lienol/openwrt-luci/tree/dev-18.06/modules/luci-mod-admin-full
> 18.06的就有......

我最后一次说，执行./scripts/feeds install -a之后才会拉取代码，才能看到那些htm文件。

trombonist852, 2020-08-12:
> 我最后一次说，执行./scripts/feeds install -a之后才会拉取代码，才能看到那些htm文件。

好的好的，大神！非常感谢，我再去操作一遍，给您添麻烦了哈！

dukh1216, 2020-09-04:
> > > > > Lienol的openwrt替换为dev-19.07，交替的luci也是19.07，为什么你都选非主流的分支啊
> > > > > https://github.com/Lienol/openwrt/blob/dev-19.07/feeds.conf.default 
> > > > > .. 。
> > > > > SRC-git的包https://github.com/Lienol/openwrt-packages.git;dev-19.07
> > > > 
> > > > 
> > > > 他的luci-19.07就没有您说的那个文件，切换到18就有了
> > > 
> > > 
> > > 有的，你需要执行这两个命令
> > > 。/scripts/feedsupdate -a 
> > > ./scripts/feeds install -a 
> > > 之后才会有
> > 
> > 
> > https://github.com/Lienol/openwrt-luci/tree/dev-19.07/modules/luci-mod-admin-full 
> > 您看看真的没有....... 
> > [https://github.com/ Lienol / openwrt-luci / tree / dev-18.06 / modules / luci-mod-admin-full](https://github.com/Lienol/openwrt-luci/tree/dev-18.06/modules/luci-mod-admin-full) 
> > 18.06的就有......
> 
> 我最后一次说，执行。/scripts/feedsinstall -a之后才会拉取代码，才能看到那些htm文件。

请问大佬有考虑一下把core补丁提交吗？如果可以合并的话会方便很多

feheadls, 2020-12-11:
> 如题请教一下各位大神，树莓派32位就可以正常显示，64位就不行，r2s也是一样，auto_core.diff的补丁也打过了，menuconfig里面autocore也选择了，可就是不显示，请大神们帮忙分析一下是什么原因？是不是还要做一些地方的改动？万分感激！！

请问menuconfig里，autocore在哪个菜单下啊，我在编译R4S固件的时候一直找不到。package文件夹下是有autocore的。

closed 2021-06-20 by coolsnowwolf