coolsnowwolf/lede #5307: 编译过程出错 提示

MiyouQ, 2020-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

编译到IPSET处出错 看提示是pkg-config有问题 试着重新安装了下还是问题依旧

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
checking for strlcpy... yes
checking pkg-config is at least version 0.9.0... pkg-config.real: error while loading shared libraries: libpkgconf.so.3: cannot open shared object file: No such file or directory
no
checking for grep that handles long lines and -e... (cached) /home/miyou/Desktop/lede/staging_dir/host/bin/grep
checking for gawk... (cached) gawk
checking for x86_64-openwrt-linux-gcc... (cached) x86_64-openwrt-linux-musl-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether x86_64-openwrt-linux-musl-gcc accepts -g... (cached) yes
checking for x86_64-openwrt-linux-musl-gcc option to accept ISO C89... (cached) none needed
checking whether x86_64-openwrt-linux-musl-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-openwrt-linux-musl-gcc... (cached) gcc3
checking whether ln -s works... yes
checking for libmnl... no
configure: error: in `/home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4':
configure: error: The pkg-config script could not be found or is too old.  Make sure it
is in your PATH or set the PKG_CONFIG environment variable to the full
path to pkg-config.

Alternatively, you may set the environment variables libmnl_CFLAGS
and libmnl_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

To get pkg-config, see <http://pkg-config.freedesktop.org/>.
See `config.log' for more details
make[3]: *** [Makefile:76: /home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/miyou/Desktop/lede/package/network/utils/ipset'
time: package/network/utils/ipset/compile#7.11#1.21#11.34
make[2]: *** [package/Makefile:113: package/network/utils/ipset/compile] Error 2
make[2]: Leaving directory '/home/miyou/Desktop/lede'
make[1]: *** [package/Makefile:107: /home/miyou/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/miyou/Desktop/lede'
make: *** [/home/miyou/Desktop/lede/include/toplevel.mk:224: world] Error 2

wushenghn, 2020-08-12:
ipset文件夹删掉重来

MiyouQ, 2020-08-12:
> ipset文件夹删掉重来

尝试了下 错误依旧

helianling, 2020-08-13:
target-x86_64_musl/linux-x86_64/ipset-7.4/ 删掉，再看一下dl文件夹中有没有ipset-7.4，有也删掉，再编译。

MiyouQ, 2020-08-13:
> target-x86_64_musl/linux-x86_64/ipset-7.4/ 删掉，再看一下dl文件夹中有没有ipset-7.4，有也删掉，再编译。

两个目录下的ipset都删掉后 从新编译过程有新下载  但是编译到ipset还是报错

`checking for closedir... yes
checking for opendir... yes
checking for readdir... yes
checking for strlcat... yes
checking for strlcpy... yes
checking pkg-config is at least version 0.9.0... pkg-config.real: error while loading shared libraries: libpkgconf.so.3: cannot open shared object file: No such file or directory
no
checking for grep that handles long lines and -e... (cached) /home/miyou/Desktop/lede/staging_dir/host/bin/grep
checking for gawk... (cached) gawk
checking for x86_64-openwrt-linux-gcc... (cached) x86_64-openwrt-linux-musl-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether x86_64-openwrt-linux-musl-gcc accepts -g... (cached) yes
checking for x86_64-openwrt-linux-musl-gcc option to accept ISO C89... (cached) none needed
checking whether x86_64-openwrt-linux-musl-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-openwrt-linux-musl-gcc... (cached) gcc3
checking whether ln -s works... yes
checking for libmnl... no
configure: error: in `/home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4':
configure: error: The pkg-config script could not be found or is too old.  Make sure it
is in your PATH or set the PKG_CONFIG environment variable to the full
path to pkg-config.

Alternatively, you may set the environment variables libmnl_CFLAGS
and libmnl_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

To get pkg-config, see <http://pkg-config.freedesktop.org/>.
See `config.log' for more details
make[3]: *** [Makefile:76: /home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/miyou/Desktop/lede/package/network/utils/ipset'
time: package/network/utils/ipset/compile#6.49#1.84#48.86
make[2]: *** [package/Makefile:113: package/network/utils/ipset/compile] Error 2
make[2]: Leaving directory '/home/miyou/Desktop/lede'
make[1]: *** [package/Makefile:107: /home/miyou/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/miyou/Desktop/lede'
make: *** [/home/miyou/Desktop/lede/include/toplevel.mk:224: world] Error 2
`

wushenghn, 2020-08-13:
明显缺少依赖没找到libmnl,安装libmnl-dev

MiyouQ, 2020-08-13:
> 明显缺少依赖没找到libmnl,安装libmnl-dev或者libmnl0 试试

安装完libmnl-dev和libmnl0后问题依旧

`checking for strlcpy... yes
checking pkg-config is at least version 0.9.0... pkg-config.real: error while loading shared libraries: libpkgconf.so.3: cannot open shared object file: No such file or directory
no
checking for grep that handles long lines and -e... (cached) /home/miyou/Desktop/lede/staging_dir/host/bin/grep
checking for gawk... (cached) gawk
checking for x86_64-openwrt-linux-gcc... (cached) x86_64-openwrt-linux-musl-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether x86_64-openwrt-linux-musl-gcc accepts -g... (cached) yes
checking for x86_64-openwrt-linux-musl-gcc option to accept ISO C89... (cached) none needed
checking whether x86_64-openwrt-linux-musl-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-openwrt-linux-musl-gcc... (cached) gcc3
checking whether ln -s works... yes
checking for libmnl... no
configure: error: in `/home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4':
configure: error: The pkg-config script could not be found or is too old.  Make sure it
is in your PATH or set the PKG_CONFIG environment variable to the full
path to pkg-config.

Alternatively, you may set the environment variables libmnl_CFLAGS
and libmnl_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

To get pkg-config, see <http://pkg-config.freedesktop.org/>.
See `config.log' for more details
make[3]: *** [Makefile:76: /home/miyou/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/ipset-7.4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/miyou/Desktop/lede/package/network/utils/ipset'
time: package/network/utils/ipset/compile#6.57#1.53#9.50
make[2]: *** [package/Makefile:113: package/network/utils/ipset/compile] Error 2
make[2]: Leaving directory '/home/miyou/Desktop/lede'
make[1]: *** [package/Makefile:107: /home/miyou/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/miyou/Desktop/lede'
make: *** [/home/miyou/Desktop/lede/include/toplevel.mk:224: world] Error 2
`

1607941875, 2021-04-27:
还没有解决吗

MiyouQ, 2021-04-28:
> 还没有解决吗

解决了 但是时间过的太久了 忘记怎么解决的了XD

closed 2021-06-20 by coolsnowwolf