coolsnowwolf/lede #5311: 是不是不支持i219-v这个网卡呢

yang05130, 2020-08-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
用的是华擎b460m-itx的主板 板载网卡是intel i219-v，试了几个固件，都不能够在openwrt里面识别到这上网口，是不是目前还没支持呢
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

ricksuzade-maker, 2020-08-13:
需要更新e1000e驱动，我前几天也在群里问，有没有方法可以升级这个驱动，目前固件里版本3.2.6，是2015年的，最新的是3.8.2.

yang05130, 2020-08-14:
> 需要更新e1000e驱动，我前几天也在群里问，有没有方法可以升级这个驱动，目前固件里版本3.2.6，是2015年的，最新的是3.8.2.

请问知道怎么了更新这个驱动吗

coolsnowwolf, 2020-08-14:
有可能是 fw 的问题，我手头没有这个网卡无法调试

yang05130, 2020-08-14:
还有没有其它的办法呢

ricksuzade-maker, 2020-08-14:
[image]
必须要更新这个版本的驱动才能支持i219v, intel 的说明里写了。

yang05130, 2020-08-14:
> [image]
> 必须要更新这个版本的驱动才能支持i219v, intel 的说明里写了。

求指导，怎么更新进去

ricksuzade-maker, 2020-08-15:
在dl文件夹里有一个叫做“linux-5.4.51.tar.xz”的文件，应该装的是linux内核，在“linux-5.4.51\drivers\net\ethernet\intel\e1000e”路径下放的e1000e驱动，不知道直接把intel官方驱动替换进去行不行，我还没试过。

closed 2021-06-20 by coolsnowwolf