coolsnowwolf/lede #5315: 最新源码刚编译能通过minidlna好像启不来了帮忙看一下

stock169, 2020-08-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：原来一直可以更新的源码编译也可以过但是minidlna启不来了，好像libjpeg.so.9缺失有关怎么解决

### (2) 路由器型号和固件版本

A：
    P&W R619AC 128M
    OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae) 

### (3) 详细日志

A：Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error loading shared library libjpeg.so.9: No such file or directory (needed by /usr/sbin/minidlnad)
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_stdio_src: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_start_compress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_destroy_decompress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_finish_compress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_CreateDecompress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_CreateCompress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_std_error: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_write_scanlines: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_set_defaults: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_read_scanlines: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_finish_decompress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_read_header: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_resync_to_restart: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_set_quality: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_start_decompress: symbol not found
Sat Aug 15 11:29:09 2020 daemon.err minidlnad[3069]: Error relocating /usr/sbin/minidlnad: jpeg_destroy_compress: symbol not found
Sat Aug 15 11:29:10 2020 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: started, v1.31.1
Sat Aug 15 11:29:10 2020 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: sending discover
Sat Aug 15 11:29:11 2020 daemon.err mpd[2968]: exception: bind to '0.0.0.0:6600' failed (continuing anyway, because binding to '[::]:6600' succeeded): Failed to bind socket: Address in use
Sat Aug 15 11:29:11 2020 daemon.err mpd[2968]: output: No 'audio_output' defined in config file
Sat Aug 15 11:29:11 2020 daemon.err mpd[2968]: output: Attempt to detect audio output device
Sat Aug 15 11:29:11 2020 daemon.err mpd[2968]: output: Attempting to detect a alsa audio device
Sat Aug 15 11:29:11 2020 daemon.err mpd[2968]: output: Successfully detected a alsa audio device
Sat Aug 15 11:29:11 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sat Aug 15 11:29:12 2020 cron.err crond[3313]: crond (busybox 1.31.1) started, log level 5
Sat Aug 15 11:29:14 2020 daemon.notice procd: /etc/rc.d/S96adbyby: udhcpc: no lease, failing
Sat Aug 15 11:29:14 2020 daemon.info dnsmasq[3568]: Connected to system UBus
Sat Aug 15 11:29:14 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led WLAN2G
Sat Aug 15 11:29:14 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led WLAN5G
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error loading shared library libjpeg.so.9: No such file or directory (needed by /usr/sbin/minidlnad)
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_stdio_src: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_start_compress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_destroy_decompress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_finish_compress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_CreateDecompress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_CreateCompress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_std_error: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_write_scanlines: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_set_defaults: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_read_scanlines: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_finish_decompress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_read_header: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_resync_to_restart: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_set_quality: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_start_decompress: symbol not found
Sat Aug 15 11:29:14 2020 daemon.err minidlnad[3656]: Error relocating /usr/sbin/minidlnad: jpeg_destroy_compress: symbol not found

stock169, 2020-08-15:
已搞定，需要加一个999-05-fix-setjmp-buffer-multiple-def.patch

hyird, 2020-08-15:
> 已搞定，需要加一个999-05-fix-setjmp-buffer-multiple-def.patch

欢迎发pr修复

brumegoo, 2020-09-07:
> 已搞定，需要加一个999-05-fix-setjmp-buffer-multiple-def.patch

咋加的？

closed 2021-06-20 by coolsnowwolf