coolsnowwolf/lede #5317: How to make builds with a 4.14 kernel

9000000, 2020-08-15:
By default make builds with the 5.4 kernel  
What should I do if I want to build 4.14 with a kernel?

Promix953, 2020-08-15:
[https://github.com/coolsnowwolf/openwrt](url)

mysteryman2045, 2020-08-15:
git checkout 20200416

9000000, 2020-08-16:
Thanks everyone

closed 2020-08-16 by 9000000

klever1988, 2020-08-16:
It's not a good way to use outdated code or branch
the perfect ways is changing the values of KERNEL_PATCHVER and KERNEL_TESTING_PATCHVER from 5.4 to 4.19
located in https://github.com/coolsnowwolf/lede/blob/master/target/linux/x86/Makefile
but 4.14 is removed from this branch

if you insist 4.14 you will have to use https://github.com/coolsnowwolf/openwrt
and change the value in https://github.com/coolsnowwolf/openwrt/blob/lede-17.01/target/linux/x86/Makefile bacause the default kernel vesion is 4.19