coolsnowwolf/lede #5319: 一开机就出现几百条系统日志记录:“daemon.notice hostapd: ACS: Survey is missing noise floor”

lfm09, 2020-08-16:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
不管是2.4G还是5G无线，只要信道设置为“auto”，路由一开机就会出现少则上百条多则几百条的重复日志记录，信息为“daemon.notice hostapd: ACS: Survey is missing noise floor”。而且，即使把日志记录等级和Cron日志级别都设置为“警告”，也还会出现这些信息。

附带反映一下LED灯的问题：HC5761A的2.4G灯即使没有无线连接也会连续闪烁，而5G灯即使有5G连接也常亮。

### (2) 路由器型号和固件版本

A：
HiWiFi HC5761A 
OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13) 

### (3) 详细日志

A：
Sun Aug 16 06:58:15 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sun Aug 16 06:58:15 2020 kern.info kernel: [   34.017982] br-lan: port 2(wlan0) entered blocking state
Sun Aug 16 06:58:15 2020 kern.info kernel: [   34.023461] br-lan: port 2(wlan0) entered disabled state
Sun Aug 16 06:58:15 2020 kern.info kernel: [   34.029251] device wlan0 entered promiscuous mode
Sun Aug 16 06:58:15 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Aug 16 06:58:15 2020 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sun Aug 16 06:58:15 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Sun Aug 16 06:58:15 2020 daemon.notice hostapd: wlan0: ACS-STARTED
Sun Aug 16 06:58:15 2020 daemon.info dnsmasq[1715]: Connected to system UBus
Sun Aug 16 06:58:15 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Sun Aug 16 06:58:18 2020 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Sun Aug 16 06:58:21 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led internet
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
[... 98 lines cut ...]
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Sun Aug 16 06:58:23 2020 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=2462 channel=11
Sun Aug 16 08:38:40 2020 daemon.info procd: - init complete -
Sun Aug 16 08:38:40 2020 kern.info kernel: [   44.384029] br-lan: port 2(wlan0) entered blocking state
Sun Aug 16 08:38:40 2020 kern.info kernel: [   44.389478] br-lan: port 2(wlan0) entered forwarding state
Sun Aug 16 08:38:40 2020 daemon.notice netifd: Network device 'wlan0' link is up
Sun Aug 16 08:38:40 2020 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Sun Aug 16 08:38:40 2020 daemon.notice hostapd: wlan0: AP-ENABLED

Aes64X, 2020-08-16:
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：
> 不管是2.4G还是5G无线，只要信道设置为“auto”，路由一开机就会出现少则上百条多则几百条的重复日志记录，信息为“daemon.notice hostapd: ACS: Survey is missing noise [...]

早就说过的，开源驱动MT76没这功能，你把信道固定就不会出现了。

lfm09, 2020-08-16:
谢谢回复！OP现在还没有MT7628的闭源驱动吧？

closed 2021-06-20 by coolsnowwolf