coolsnowwolf/lede #5323: 小米路由器3 Pro在升级到5.4内核后无法使用PPPoE拨号

hcwang, 2020-08-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前使用coolsnowwolf/openwrt (内核版本为4.14)时，可以完美使用pppoe拨号，但升级到coolsnowwolf/lede(内核版本为5.4)时，无法建立pppoe拨号。是否少选了某些模块？

### (2) 路由器型号和固件版本

A：小米路由器拔号

### (3) 详细日志

A：无

hcwang, 2020-08-17:
所使用的config文件如下：　https://github.com/hcwang/openwrt-xiaomir3p/blob/master/xiaomi_r3p

AmadeusGhost, 2020-08-18:
不给系统日志那没法解决

hcwang, 2020-08-20:
> 不给系统日志那没法解决

由于无法拔号，所以当时回滚到4.1.4了，为了找日志，今早重新刷了一下，日志如下：

'''
0x00000000-0x000fffff 64bit]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.243992] pci 0000:02:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:01.0 (capable of 4.000 Gb/s with 5 GT/s x1 link)
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.259086] pci 0000:00:01.0: PCI bridge to [bus 02-ff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.264320] pci 0000:00:01.0:   bridge window [io  0x0000-0x0fff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.270407] pci 0000:00:01.0:   bridge window [mem 0x60100000-0x601fffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.277195] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.283852] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.290459] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.297415] pci 0000:00:01.0: BAR 0: no space for [mem size 0x80000000]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.304024] pci 0000:00:01.0: BAR 0: failed to assign [mem size 0x80000000]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.310981] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.317762] pci 0000:00:01.0: BAR 8: assigned [mem 0x60100000-0x601fffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.324547] pci 0000:00:00.0: BAR 1: assigned [mem 0x60200000-0x6020ffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.331343] pci 0000:00:01.0: BAR 1: assigned [mem 0x60210000-0x6021ffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.338127] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.344912] pci 0000:00:01.0: BAR 7: assigned [io  0x1e161000-0x1e161fff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.351704] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff 64bit]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.359016] pci 0000:00:00.0: PCI bridge to [bus 01]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.363980] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.370763] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.377551] pci 0000:02:00.0: BAR 0: assigned [mem 0x60100000-0x601fffff 64bit]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.384860] pci 0000:00:01.0: PCI bridge to [bus 02]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.389823] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.396606] pci 0000:00:01.0:   bridge window [mem 0x60100000-0x601fffff]
Thu Aug 20 02:00:32 2020 kern.info kernel: [    5.403921] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control off
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    5.405066] UBI: auto-attach mtd9
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    5.414141] ubi0: attaching mtd9
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    7.958299] ubi0: scanning is finished
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    7.981253] ubi0: attached mtd9 (name "ubi", size 245 MiB)
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    7.986769] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    7.993624] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.000394] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.007337] ubi0: good PEBs: 1964, bad PEBs: 0, corrupted PEBs: 0
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.013413] ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.020619] ubi0: max/mean erase counter: 48/32, WL threshold: 4096, image sequence number: 1984940186
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.029896] ubi0: available PEBs: 0, total reserved PEBs: 1964, PEBs reserved for bad PEB handling: 40
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.039199] ubi0: background thread "ubi_bgt0d" started, PID 462
Thu Aug 20 02:00:32 2020 kern.info kernel: [    8.041162] block ubiblock0_0: created from ubi0:0(rootfs)
Thu Aug 20 02:00:32 2020 kern.notice kernel: [    8.050689] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
Thu Aug 20 02:00:32 2020 kern.info kernel: [    8.057645] hctosys: unable to open rtc device (rtc0)
Thu Aug 20 02:00:32 2020 kern.info kernel: [    8.069768] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
Thu Aug 20 02:00:32 2020 kern.info kernel: [    8.081448] Freeing unused kernel memory: 1296K
Thu Aug 20 02:00:32 2020 kern.warn kernel: [    8.086029] This architecture does not have kernel memory protection.
Thu Aug 20 02:00:32 2020 kern.info kernel: [    8.092461] Run /sbin/init as init process
Thu Aug 20 02:00:32 2020 user.info kernel: [    8.698908] init: Console is alive
Thu Aug 20 02:00:32 2020 user.info kernel: [    8.702555] init: - watchdog -
Thu Aug 20 02:00:32 2020 user.info kernel: [   11.124623] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[... cut ...]
Thu Aug 20 02:01:53 2020 daemon.notice netifd: wan (8766): sh: out of range
Thu Aug 20 02:01:53 2020 daemon.info pppd[8801]: Plugin rp-pppoe.so loaded.
Thu Aug 20 02:01:53 2020 daemon.info pppd[8801]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Thu Aug 20 02:01:53 2020 daemon.notice pppd[8801]: pppd 2.4.8 started by root, uid 0
Thu Aug 20 02:01:54 2020 daemon.warn odhcpd[2262]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

'''

AmadeusGhost, 2020-08-20:
```
daemon.warn pppd[5514]: Timeout waiting for PADO packets
daemon.err pppd[5514]: Unable to complete PPPoE Discovery
daemon.info pppd[5514]: Exit.
daemon.notice netifd: Interface 'wan' is now down
daemon.notice netifd: Interface 'wan' is setting up now
daemon.err insmod: module is already loaded - slhc
daemon.err insmod: module is already loaded - ppp_generic
daemon.err insmod: module is already loaded - pppox
daemon.err insmod: module is already loaded - pppoe
daemon.notice netifd: wan (6790): uci: Entry not found
daemon.notice netifd: wan (6790): sh: out of range
daemon.info pppd[6812]: Plugin rp-pppoe.so loaded.
daemon.info pppd[6812]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
daemon.notice pppd[6812]: pppd 2.4.8 started by root, uid 0
daemon.warn odhcpd[2262]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```
问题部分在这

hcwang, 2020-08-20:
> ```
> daemon.warn pppd[5514]: Timeout waiting for PADO packets
> daemon.err pppd[5514]: Unable to complete PPPoE Discovery
> daemon.info pppd[5514]: Exit.
> daemon.notice netifd: Interface 'wan' is now down [...]

谢谢！不过抱歉，恕我愚钝，应该如何修复呢？如果改成4.1.4就完全正常了。

AmadeusGhost, 2020-08-20:
> 谢谢！不过抱歉，恕我愚钝，应该如何修复呢？如果改成4.1.4就完全正常了。

我也不知道诶，我去问问大佬

hcwang, 2020-08-21:
> > 谢谢！不过抱歉，恕我愚钝，应该如何修复呢？如果改成4.1.4就完全正常了。
> 
> 我也不知道诶，我去问问大佬

多谢。我在网上查找到有两个原因：

1.　局端问题：说是服务器方（我这边是联通）的配置错误，需要重新配置，但我应该不是这个问题，因为使用之前的4.1.4是正常的；
2.　WAN的MAC地址克隆问题：说是WAN的MAC地址是克隆的，但我应该也不是此问题，因为没有克隆呀。

closed 2021-06-20 by coolsnowwolf