coolsnowwolf/lede #5324: Newifi D2开源WiFi驱动只有WEP加密

Cathgao, 2020-08-17:
## 1.关于你要提交的问题

新3使用开源驱动，刷入后WiFi只能选择无加密和WEP加密

[image]

## 2. 详细叙述
###  路由器型号和固件版本

```
型号 | Newifi-D2
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13)
内核版本 | 5.4.51
```

### 配置文件：
```
#
# 机型选择
#
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y

#
# IPV6支持
#
CONFIG_IPV6=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ip6tables-extra=y
CONFIG_PACKAGE_6in4=y

#
# samba扩展
#
CONFIG_PACKAGE_autosamba=y
#CONFIG_PACKAGE_samba36-server is not set

#
# USB与存储相关支持
#
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_kmod-fs-qntfs=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_ntfs-3g=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-fuse=y
CONFIG_PACKAGE_kmod-mtd-rw=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_kmod-sdhci=y
CONFIG_PACKAGE_kmod-mmc-spi=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-dwc3=y
CONFIG_PACKAGE_kmod-usb-dwc3-of-simple=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-extras=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-usb3=y

#
# 无线驱动
#
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_MAC80211_DEBUGFS=y
CONFIG_PACKAGE_MAC80211_MESH=y
#use opensource wifi driver
CONFIG_PACKAGE_kmod-mt7603=y
CONFIG_PACKAGE_kmod-mt7603e=n
CONFIG_PACKAGE_kmod-mt76x2=y
CONFIG_PACKAGE_kmod-mt76x2-common=y
CONFIG_PACKAGE_kmod-mt76x2e=n

#
# 插件集成
#
#CONFIG_PACKAGE_luci-app-accesscontrol is not set
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-aria2=n
CONFIG_PACKAGE_luci-app-arpbind=y
CONFIG_PACKAGE_luci-app-autoreboot=n
CONFIG_PACKAGE_luci-app-cifsd=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-dnspod=n
CONFIG_PACKAGE_luci-app-dnsforwarder=n
CONFIG_PACKAGE_luci-app-filetransfer=n
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-flowoffload=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-ipsec-vpnd=n
CONFIG_PACKAGE_luci-app-mwan3=n
CONFIG_PACKAGE_luci-app-mwan3helper=n
CONFIG_PACKAGE_luci-app-nlbwmon=n
CONFIG_PACKAGE_luci-app-pptp-server=n
CONFIG_PACKAGE_luci-app-ramfree=y
#CONFIG_PACKAGE_luci-app-samba is not set
CONFIG_PACKAGE_luci-app-serverchan=y
CONFIG_PACKAGE_luci-app-sqm=n
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=n
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=n
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=n
CONFIG_PACKAGE_luci-app-syncdial=n
CONFIG_PACKAGE_luci-app-transmission=n
CONFIG_PACKAGE_luci-app-ttyd=n
CONFIG_PACKAGE_luci-app-uhttpd=y
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-app-v2ray-server=n
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-app-wifischedule=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-app-xlnetacc=n
CONFIG_PACKAGE_luci-app-zerotier=n
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-mtwifi=n
# CONFIG_PACKAGE_qBittorrent is not set
# CONFIG_DEFAULT_luci-app-qbittorrent is not set

#
# 主题
#
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-bootstrap=n

#
# 基础命令及其他
#
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_lscpu=y
CONFIG_PACKAGE_mount-utils=y
```

Cathgao, 2020-08-17:
正常来说 `wpa_supplicant`不是默认勾选的吗？

Cathgao, 2020-08-17:
选中这两个模块之后变正常了
```
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wpa-supplicant=y
```
为什么开源驱动没有默认勾选？

AmadeusGhost, 2020-08-18:
wpad-openssl 是在 device package 默认勾选的

Cathgao, 2020-08-18:
> wpad-openssl 是在 device package 默认勾选的

那对比原配置我也只是取消选中了这三项，原配置是没有问题的，取消这三个之后wpa就自己没了
```
CONFIG_PACKAGE_luci-app-mtwifi=n
CONFIG_PACKAGE_kmod-mt7603e=n
CONFIG_PACKAGE_kmod-mt76x2e=n
```

jjm2473, 2020-08-25:
这个跟驱动没有关系，只要选择wpad-openssl就行了，里面包括了hostapd和wpa-supplicant

DongfeiSay, 2020-08-25:
wpad-openssl这东西藏哪了。。。找了半天没找到

Cathgao, 2020-08-28:
> 这个跟驱动没有关系，只要选择wpad-openssl就行了，里面包括了hostapd和wpa-supplicant

疑惑的就是取消闭源驱动选中开源驱动后，wpad-openssl默认被取消了

jjm2473, 2020-08-28:
这是依赖没设置好吧

closed 2021-06-20 by coolsnowwolf