coolsnowwolf/lede #5327: 关于aes-xxx-gcm的提问

kl9800, 2020-08-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：自身水平有限，懒得找源码了，所以干脆这里问问，上网看到一个大神说，现在的V2与饮料的AES，都是用的软加解密，没有调用CPU指令集的代码？那么有没有必要专门购入支持AES的CPU的软路由？
    如果都是软加解密，岂不是没必要？

### (2) 路由器型号和固件版本

A：XXXX

### (3) 详细日志

A：xxxx

hcym, 2020-08-18:
软硬搞不懂，这个得大雕们解答，不过机器性能还是很重要，我现在根本就没法用硬路由，都是单臂加ap，尽管是移动100，看电视YouTube时明显5020u比蜗牛n1好多了，手机端没什么感觉，估计用k3k2也可以

Cathgao, 2020-08-18:
你怕AES影响性能就用chacha20

Aes64X, 2020-08-18:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x ] 没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：自身水平有限，懒得找源码了，所以干脆这里问问，上网看到一个大神说，现在的V2与饮料的AES，都是用的软加解密，没有调用CPU指令集的代码？那么有没有必要专门购入支持AES的CPU的软路由？
> 如果都是软加解密，岂不是没必要？
> 
> ### (2) 路由器型号和固件版本
> A：XXXX
> 
> ### (3) 详细日志
> A：xxxx

原本就没必要，这话题聊得多了就是JS接手连油管主播开炒。
为什么没必要？
假如以流畅起飞4K为标准，MT7621就可以很好地完成（QCA9563或7620之流请扔掉），大多数情况下不是路由性能不够，而是飞机航线太垃圾。
那么，仅仅是流畅4K，又不是科研学术需求，定位好自己的需求。
大流量起飞需求的，航线往往自己画，能力水平也是够的。

（附：N1的SOC带有AES硬件加解密，R2S和树莓派4的BCM2711没有，但其本身性能已远超4K起飞需求）
（MT7621 SOC 本身带有EIP93硬件加解密，目前的代码中已运行但没有更详细的测试了）

kl9800, 2020-08-18:
谢谢！

closed 2020-08-18 by kl9800