coolsnowwolf/lede #5335: 在启动SSR plus+后dnsmasq crash导致域名无法解析。

raiuka, 2020-08-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：全新安装的X86 LEDE运行在ESXI上，当启动55R的过程中发现dnsmasq crash，导致域名无法解析，后发现dnsmasq启动失败。关闭SSR plus+后问题消失。

### (2) 路由器型号和固件版本

A：OpenWrt / LuCI Master (git-20.223.26773-d18ef13)

### (3) 详细日志

A：
系统日志：

Thu Aug 20 00:50:00 2020 daemon.crit dnsmasq[14021]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:00 2020 daemon.crit dnsmasq[14021]: FAILED to start up
Thu Aug 20 00:50:05 2020 daemon.crit dnsmasq[14094]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:05 2020 daemon.crit dnsmasq[14094]: FAILED to start up
Thu Aug 20 00:50:10 2020 daemon.crit dnsmasq[14117]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:10 2020 daemon.crit dnsmasq[14117]: FAILED to start up
Thu Aug 20 00:50:15 2020 daemon.crit dnsmasq[14143]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:15 2020 daemon.crit dnsmasq[14143]: FAILED to start up
Thu Aug 20 00:50:20 2020 daemon.crit dnsmasq[14161]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:20 2020 daemon.crit dnsmasq[14161]: FAILED to start up
Thu Aug 20 00:50:25 2020 daemon.crit dnsmasq[14184]: recompile with HAVE_IPSET defined to enable ipset directives at line 1 of /tmp/dnsmasq.ssr/gfw_base.conf
Thu Aug 20 00:50:25 2020 daemon.crit dnsmasq[14184]: FAILED to start up
Thu Aug 20 00:50:25 2020 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Thu Aug 20 00:50:37 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Aug 20 00:50:37 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Aug 20 00:50:37 2020 daemon.info dnsmasq[14405]: Connected to system UBus
Thu Aug 20 00:50:37 2020 cron.err crond[14439]: crond (busybox 1.31.1) started, log level 5
Thu Aug 20 00:50:37 2020 daemon.info dnsmasq[14515]: Connected to system UBus
Thu Aug 20 00:52:37 2020 cron.err crond[21090]: crond (busybox 1.31.1) started, log level 5
Thu Aug 20 00:52:37 2020 cron.err crond[21200]: crond (busybox 1.31.1) started, log level 5

请问这个问题我该从哪里排查？

nivalxer, 2020-08-20:
尝试更新一下GFW列表

raiuka, 2020-08-20:
> 尝试更新一下GFW列表

已经更新过了。。。还是会crash掉，不管是GFW模式还是代理模式还是出国模式都会crash掉

coolsnowwolf, 2020-08-21:
DNSMASQ 没有选上 IPSET 的支持

closed 2021-06-20 by coolsnowwolf