coolsnowwolf/lede #5352: 请问这是什么问题 mac format error: :  ## mac num = 0

authon, 2020-08-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：启动出现这个错误

### (2) 路由器型号和固件版本

A：OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13)

### (3) 详细日志

A：[    0.000000] Linux version 5.4.51 (runner@fv-az195) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-46524d9)) #0 SMP Mon Aug 17 17:41:42 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=a16a0916-fda4-0be8-ba4c-cfdf3c25ac02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bffddfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bffde000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.13.0-0-gf21b5a4aeb02-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 23d5001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 2954925861868 cycles
[    0.000005] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000010] tsc: Detected 2294.674 MHz processor
[    0.002375] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002377] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002381] last_pfn = 0x140000 max_arch_pfn = 0x400000000
[    0.002409] MTRR default type: write-back
[    0.002410] MTRR fixed ranges enabled:
[    0.002411]   00000-9FFFF write-back
[    0.002412]   A0000-BFFFF uncachable
[    0.002412]   C0000-FFFFF write-protect
[    0.002413] MTRR variable ranges enabled:
[    0.002414]   0 base 00C0000000 mask FFC0000000 uncachable
[    0.002415]   1 disabled
[    0.002415]   2 disabled
[    0.002415]   3 disabled
[    0.002416]   4 disabled
[    0.002416]   5 disabled
[    0.002417]   6 disabled
[    0.002417]   7 disabled
[    0.012792] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.022938] last_pfn = 0xbffde max_arch_pfn = 0x400000000
[    0.033681] found SMP MP-table at [mem 0x000f5a80-0x000f5a8f]
[    0.033769] Using GB pages for direct mapping
[    0.034061] ACPI: Early table checksum verification disabled
[    0.034084] ACPI: RSDP 0x00000000000F5880 000014 (v00 BOCHS )
[    0.034092] ACPI: RSDT 0x00000000BFFE15A0 000034 (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.034100] ACPI: FACP 0x00000000BFFE139A 000074 (v01 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.034106] ACPI: DSDT 0x00000000BFFDF040 00235A (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[... 592 lines cut ...]
[   10.849409] br-lan: port 1(eth1) entered disabled state
[   10.851566] device eth1 entered promiscuous mode
[   11.010740] 8021q: adding VLAN 0 to HW filter on device eth0
[   11.045406] fast-classifier (PBR safe v2.1.4a): starting up
[   11.047418] fast-classifier: registered
[   11.970768] config size: 25,data = { "op": 3, "data": { } }
[   11.970768] 
[   11.972887] #########show app status##########
[   11.974063] 
[   11.974063] 
[   11.974063] 
[   11.991844] config size: 45,data = { "op": 4, "data": { "mac_list": [ ":" ] } }
[   11.991844] 
[   11.993921] mac format error: :
[   11.994833] ## mac num = 0
[   11.995678] #########show app status##########
[   11.996793] 
[   11.996793] 
[   11.996793] 
[   13.690066] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   13.858957] br-lan: port 1(eth1) entered blocking state
[   13.860055] br-lan: port 1(eth1) entered forwarding state
[   14.022440] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   14.024406] igb 0000:00:11.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   19.312754] pppoe-wan: renamed from ppp0

lxwq, 2020-08-25:
同样有这个错误提示,不知道哪的原因

sirpdboy, 2020-08-25:
经测试，这个是 安装 应用过滤 来的。

authon, 2020-08-25:
> 经测试，这个是 安装 应用过滤 来的。

好的，我试试去掉，请问你是怎么测试出来的呢？

closed 2021-06-20 by coolsnowwolf