coolsnowwolf/lede #5353: 分享已解决overlay挂载失败问题的方法

mysteryman2045, 2020-08-24:
分享已解决overlay挂载失败问题的方法

在https://bugs.openwrt.org/index.php?do=details&task_id=2231  上看到了讨论解决overlay失败的内容。从去年4月一直到今年8月，不同人在给出不同的解决方案。可惜全是英文，完全啃不动。
前段时间看恩山上dogbutcat说最后一个叫Qi Liu的给出的最新的patch，已经解决overlay挂载问题，于是自己也尝试一下。

liuqi's patch about fstools 
https://patchwork.ozlabs.org/project/openwrt/patch/BYAPR03MB444052C4846ECA6DB88B098D955E0@BYAPR03MB4440.namprd03.prod.outlook.com/ https://patchwork.ozlabs.org/project/openwrt/patch/BYAPR03MB4440353BF13FCB2C80389186955E0@BYAPR03MB4440.namprd03.prod.outlook.com/

这两个patch分别修改package/system/fstools和build_dir/target-x86_64_musl/fstools-2020-***目录下的对应文件，由于没有看到patches文件夹，那就应该是没有历史patch，所以我就直接在原文件上修改，没有再去做patch。

fstools我用的是官方7.11的版本，自己改的Makefile里的HASH，DATE，VERSION。

如果本地第一次编译，没有build_dir/target-x86_64_musl/fstools-2020-07-11-53453438目录，
分别执行make package/system/fstools/download 和 make package/system/fstools/prepare，就有了。
一定要先make menuconfig，生成.config文件。

最后确实在x86-64下实现了成功挂载overlay，于是将修改好的对应文件备份，上传到github。
Actions里添加colne自己上传文件和替换package/system/fstools/Makefile的步骤。
添加make package/system/fstools/download 和 make package/system/fstools/prepare，
获取到build_dir/target-x86_64_musl/fstools-2020-07-11-53453438目录，继续替换修改后的文件。
有了这几个步骤，最后在Actions下编译出的包也是OK的。

另外，还是要请大家帮忙看看，我这里直接按照patch内容修改对应文件，会不会有其他我没有考虑到的问题。
fstools我只是修改了Makefile，其他文件貌似没有变动，所以没有修改，会不会也有其他我没有考虑到的问题。
谢谢！

mysteryman2045, 2020-08-24:
更正一下，直接make package/system/fstools/download 和 make package/system/fstools/prepare，
获取到build_dir/target-x86_64_musl/fstools-2020-07-11-53453438目录，然后替换修改后的文件。
编译后并不能达到修复overlay挂载失败的问题，但是第一次make -j$(nproc) V=s结束后，再替换文件。
再执行一次make -j$(nproc) V=s，编译后的固件包是可以解决overlay挂载失败的问题的。
能看懂的，能不能说明一下，问题出在哪？该怎么解决，可以一次编译就OK。
第二次编译虽然也就20分钟左右，但是看上去有点傻。

Gokou-Ruri, 2020-08-25:
MT7620/7621路由器能成功挂载，挂载失败的时候可以检查看看是不是 /mnt/`磁盘名`/etc/.extroot-uuid 没删除？一般在重新刷机或者更换overlay分区的磁盘以后会碰上这个问题。

feverrun, 2020-09-05:
我也是怎么挂载都失败， 算了换个系统

mysteryman2045, 2020-09-05:
> 我也是怎么挂载都失败， 算了换个系统

我试过自己编译lean和Lienol，还有官方的openwrt，不同内核版本，不同tag，都存在overlay挂载失败的问题。所以换系统也不一定能解决。
目前也就试过用esir的包没有问题。
还有就是按照我上边说的，用上边的两个补丁就能搞定。
还有按照你楼上的说法试一试，因为我用补丁之后的效果跟他一样，/etc/.extroot-uuid 是会被删除掉的。

feverrun, 2020-09-08:
官方的这个版本openwrt-19.07.3-x86-64-combined-ext4.img我试过，不会存在挂载失败的问题，lede编译ext4结尾的固件应该也不会有问题，我把openwrt运行在虚拟机上，可以做快照不怕文件损坏，对什么格式不敏感

mysteryman2045, 2020-09-19:
给esir赞一个！！！
esir已经在9月16号把我提到的这两个补丁提交到项目里了，以后大家直接编译打包出来就不用担心overlay挂载问题了。

closed 2020-09-19 by mysteryman2045