coolsnowwolf/lede #5356: passwall界面上提示尚无任何配置

madman43, 2020-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
[https://www.right.com.cn/forum/data/attachment/forum/202008/25/102428itbhbbtbwvb7bhwn.png](url)
[https://www.right.com.cn/forum/data/attachment/forum/202008/25/102428cyx2l2do9z92bxkx.png](url)
[https://www.right.com.cn/forum/data/attachment/forum/202008/25/102715onu1unbyz6xg15o6.png](url)
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
好像是luci bug？我重启后还是无法显示，清除浏览器缓存也是无法显示，突然就消失了
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
N1黑色版 内核： 5.7.15 
Openwrt: R20.7.20 by lean & lienol (截止2020.08.15下午的最新源码）。
A：

### (3) 详细日志

A：

yubanmeiqin9048, 2020-08-25:
反馈问题前先弄清楚该到什么地方反馈，PASSWALL不是lean做的

madman43, 2020-08-25:
> 反馈问题前先弄清楚该到什么地方反馈，PASSWALL不是lean做的

okok

xx10n31y, 2020-08-25:
反馈passwall的问题麻烦去隔壁的Lionel发issue

closed 2020-08-26 by coolsnowwolf

ufo22940268, 2023-01-26:
碰到了同样的问题，~~虽然知道这个问题问错地方了，但是因为这里google上排名最靠前~~

我的解决方法是

1. 删除luci-app-passwall
1. `rm /etc/config/passwall*`
2. 重新安装luci-app-passwall

CG1995, 2024-01-22:
> 碰到了同样的问题(~虽然知道这个问题问错地方了，但是因为这里google上排名最靠前~)，我的解决方法是
> 
> 1. 删除luci-app-passwall
> 2. `rm /etc/config/passwall*`
> 3. 重新安装luci-app-passwall

谢谢你，解决了