coolsnowwolf/lede #5357: Netflix分流会突然不工作

BGzErol, 2020-08-25:
日志：
Tue Aug 25 15:23:16 2020 daemon.err nlbwmon[3624]: Netlink receive failure: Out of memory
Tue Aug 25 15:23:16 2020 daemon.err nlbwmon[3624]: Unable to dump conntrack: No buffer space available
Tue Aug 25 16:51:04 2020 daemon.err nlbwmon[3624]: Netlink receive failure: Object busy
Tue Aug 25 16:51:04 2020 daemon.err nlbwmon[3624]: Unable to dump conntrack: I/O error

有时会突然报出Out of memory的错误然后分流罢工，且不会随着时间推移自行恢复，此时内存可用数为1.8G
180-1的时候我记得也有，目前用的固件版本为8月12日的版本

BGzErol, 2020-08-25:
似乎我看错日志了，把Netlin看出Netflix了，不过Netflix分流的问题仍然依旧，不知道为何会突然不工作，得重启插件才可😂

swxk521, 2020-08-25:
是不是故障切换的服务器和奈飞分流设置的一样，就会出现奈飞不能上去问题

BGzErol, 2020-08-25:
> 是不是故障切换的服务器和奈飞分流设置的一样，就会出现奈飞不能上去问题

我故障切换功能是关着的

BGzErol, 2020-08-27:
对了，当Netflix分流不工作的时候打开Netflix.com会显示无法找到dns，怀疑是pdnsd的原因，再翻到了以前L大添加dns2socks的有关资料，似乎说dns2socks对Netflix分流要更好，不知道切换后是否还会出现分流不工作的问题

coolsnowwolf, 2020-09-13:
已修复

closed 2021-06-20 by coolsnowwolf