coolsnowwolf/lede #5358: uefi－gtp的固件去哪了？

doumeki, 2020-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ＮＯ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
４.x 的版本编译出来的固件名称是：openwrt-x86-64-uefi-gpt-squashfs
５.x 的版本编译出来的固件名称是：openwrt-x86-64-generic-squashfs-combined-efi
两个都是efi的固件，但是5.x去很诡异地使用了combined 这个词（按之前的理解这个应该是MBR）
使用磁盘精灵4.x的打开没问题，５.x的打开说GPT需要修正。
[image]

使用fdisk查看：
4.x的

Device       Start      End  Sectors  Size Type
/dev/sda1      512    33279    32768   16M Linux filesystem
/dev/sda2    33792    35839     2048    1M EFI System
/dev/sda3    36352    38399     2048    1M BIOS boot
/dev/sda4    38912  4233215  4194304    2G Linux filesystem

5.x 的
Device      Start    End Sectors  Size Type
/dev/sda1     512  33279   32768   16M EFI System
/dev/sda2   33792 648191  614400  300M Microsoft basic data
/dev/sda128    34    511     478  239K BIOS boot

感觉像是自己定义的格式，5.x的能否按以前的磁盘格式进行打包？

zxc131415, 2020-08-26:
新的可以用啊

Pastechn, 2020-09-04:
的确会提示分区表有问题 但是不影响使用

closed 2021-06-20 by coolsnowwolf