coolsnowwolf/lede #5359: nat回环wenti

BlackMask007, 2020-08-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有类似的issue，但没有解决办法，issue已被close。

## 2. 详细叙述
设置动态域名解析后局域网内无法用域名访问局域网内的主机。

### (1) 具体问题
局域网里面访问动态域名加端口打不开内网服务，必须要内网ip才行。

A：

### (2) 路由器型号和固件版本
固件版本 | OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13)
内核版本 | 5.4.51

mcucore, 2020-08-26:
前2天编译的X86，同版本号，没发现这问题。

hxlls, 2020-08-26:
请关闭Flow offload,或者转用SFE,如果编译了docker 请重新编译没docker 的版本,或者在docker 启用macvlan 进行桥接,也许能解决这个问题

BlackMask007, 2020-08-26:
> 请关闭Flow offload,或者转用SFE,如果编译了docker 请重新编译没docker 的版本,或者在docker 启用macvlan 进行桥接,也许能解决这个问题,还有,大雕早就已经默认是SFE了,如果用大雕源码,不自行改动这个选项,应该不存在这个问题,如果是在其他源码上合并大雕的源码,请向对应的地方提交问题

请问在哪里关闭Flow offload？
我没有找到相关选项。

FusionPlmH, 2020-08-27:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ ]  没有类似的issue
>   有类似的issue，但没有解决办法，issue已被close。
>
> ## 2. 详细叙述
> 设置动态域名解析后局域网内无法用域名访问局域网内的主机。
>
> ### (1) 具体问题 [...]

确实我经常发现我port forwarding 内网其他ip端口，透过ddns经常访问不了，就算是外网ip也不行。

通常要做dmz 映射几个机器全部端口才行

mysteryman2045, 2020-08-27:
> > 请关闭Flow offload,或者转用SFE,如果编译了docker 请重新编译没docker 的版本,或者在docker 启用macvlan 进行桥接,也许能解决这个问题,还有,大雕早就已经默认是SFE了,如果用大雕源码,不自行改动这个选项,应该不存在这个问题,如果是在其他源码上合并大雕的源码,请向对应的地方提交问题
> 
> 请问在哪里关闭Flow offload？
> 我没有找到相关选项。

LUCI里边

Dimitri2020007, 2020-08-29:
LuCI网页里的turbo acc有一个是flowoffload

BlackMask007, 2020-08-30:
> LuCI网页里的turbo acc有一个是flowoffload

首先感谢你的回复，但我在turbo acc页面中没有找到flowoffload

只有如下设置：
SFE 加速状态 | 运行中
BBR 加速 | 运行中
FULLCONE NAT加速 | 运行中
DNS 加速

FusionPlmH, 2020-08-30:
> > LuCI网页里的turbo acc有一个是flowoffload
> 
> 首先感谢你的回复，但我在turbo acc页面中没有找到flowoffload
> 
> 只有如下设置：
> SFE 加速状态 | 运行中
> BBR 加速 | 运行中
> FULLCONE NAT加速 | 运行中
> DNS 加速

sfe就是

Dimitri2020007, 2020-08-31:
关了sfe再试试

closed 2021-06-20 by coolsnowwolf