coolsnowwolf/lede #5360: SSR+ Netflix分流模式下域名白名单无效

lynnial, 2020-08-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：NAS架设了Plex服务器并开启外网访问，SSR+大陆白名单模式，添加plex.tv和plex.direct这两个域名到白名单以后Plex可以正确获取wan口ip，但只要SSR+一开启奈飞分流Plex就会获取到代理ip地址导致外网无法访问Plex。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

swxk521, 2020-08-26:
抓包看看，我群晖之前也是，之后quickconnect.to quickconnect.cn synology.com synology.cn，这几个设置后就没问题了

lynnial, 2020-08-27:
> 抓包看看，我群晖之前也是，之后quickconnect.to quickconnect.cn synology.com synology.cn，这几个设置后就没问题了

关键是我把NAS IP加到白名单PLEX还是获取代理IP。。。软路由走的旁路模式，现在把NAS网关手工设置到主路由顶着用了

coolsnowwolf, 2020-08-27:
可以把白名单贴上来看看吗

lynnial, 2020-08-27:
> 可以把白名单贴上来看看吗

白名单里面放了两个域名plex.tv和plex.direct，如果不开启奈飞分流模式，PLEX服务器获取的是正确的IP，但开启奈飞分流后PLEX服务器就会获取到代理的IP地址。同样还测试过把NAS IP加入不走代理名单，只要不开奈飞分流模式也是可以获取正确IP的。SSR+是180-5版本。

jamesbean08, 2020-08-27:
这个问题我也碰到了，个人推断和aws有关，可能plex的远程访问功能使用了亚马逊云服务。

我测试过在gfw模式下，如果ssr+的netflix分流ip段采用默认的netflix+aws，plex的远程访问页面会读出netflix分流服务器的ip地址；如果改用netflix only，plex可以读出正确的本地wan口ip。

供参考。

lynnial, 2020-08-27:
> 这个问题我也碰到了，个人推断和aws有关，可能plex的远程访问功能使用了亚马逊云服务。
> 
> 我测试过在gfw模式下，如果ssr+的netflix分流ip段采用默认的netflix+aws，plex的远程访问页面会读出netflix分流服务器的ip地址；如果改用netflix only，plex可以读出正确的本地wan口ip。
> 
> 供参考。

我有空试下，多谢。

closed 2021-06-20 by coolsnowwolf