coolsnowwolf/lede #5363: 重启SSR plus+有时无法运行

sky2016cn, 2020-08-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装的x86-64，断电重启，SSR plus+有时无法运行，需要点击客户端的“保存&应用”，才能正常。
B：“备份与升级”里升级固件时，有时上传的固件会选择挂载的硬盘，重启一下才能正确识别系统盘。

### (2) 路由器型号和固件版本

A：x86-64   OpenWrt R20.7.20 / LuCI Master (git-20.191.36863-eee6bae)

### (3) 详细日志

A：
SSR plus+日志
------------end------------
2020-08-27 10:37:19 Netflix shunt:V2Ray 4.27.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
2020-08-27 10:37:18 Main node:V2Ray 4.27.0 (OpenWrt) Lean (go1.14.4 linux/amd64) Started!
-----------start-----------

lexin8, 2020-08-27:
Switch 4.14.193 kernel latest source code compilation

sky2016cn, 2020-08-27:
是要把ubuntu的内核版本更改为Linux4.14.193内核版本？

hcym, 2020-08-27:
最近x86一直用的官方版，armv8断断续续基本都是编译的大雕版，很稳定

mysteryman2045, 2020-08-27:
> 是要把ubuntu的内核版本更改为Linux4.14.193内核版本？

怎么可能会是Ubuntu的内核。你正常编译x86，现在内核就是5.4的版本，除非自己去 target/linux/x86下改Makefile。不会跟内核版本有关的，否则所有人都有问题。更不会是4.14.193的内核。target/linux/x86下也就剩下5.4和4.19内核的编译配置了。

closed 2021-06-20 by coolsnowwolf