coolsnowwolf/lede #5366: 严格按照readme中的步骤执行命令，编译失败。

huluwa-dev, 2020-08-27:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：在Debian 10上 编译x86版本的lege，最新版本

### (3) 详细日志

A：
touch /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.prepared_9e5d5130a79e311ef0dd43b67302e618_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.configured_*
rm -f /home/eric/lede/staging_dir/target-x86_64_musl/stamp/.default-settings_installed
touch /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.built
touch /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.built_check
touch /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.built
rm -rf /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings.installed /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings
mkdir -p /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings
install -d -m0755 /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings/etc/uci-defaults
install -m0755 ./files/zzz-default-settings /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings/etc/uci-defaults/99-default-settings
install -d -m0755 /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings/usr/lib/lua/luci/i18n
po2lmo ./i18n/default.zh-cn.po /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings/usr/lib/lua/luci/i18n/default.zh-cn.lmo
bash: po2lmo: command not found
make[3]: *** [Makefile:49: /home/eric/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings.installed] Error 127
make[3]: Leaving directory '/home/eric/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.19#0.06#0.25
make[2]: *** [package/Makefile:113: package/lean/default-settings/compile] Error 2
make[2]: Leaving directory '/home/eric/lede'
make[1]: *** [package/Makefile:107: /home/eric/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/eric/lede'
make: *** [/home/eric/lede/include/toplevel.mk:224：world] 错误 2

bin20088, 2020-08-27:
bash: po2lmo: command not found

huluwa-dev, 2020-08-27:
找到解决办法了：需要在最后一个编译命令前，需要单独编译一下po2lmo，命令如下：
make package/feeds/luci/luci-base/compile V=99

closed 2020-08-27 by huluwa-dev

zhubinsheng, 2021-09-08:
> make package/feeds/luci/luci-base/compile V=99

 *** No rule to make target 'package/feeds/luci/luci-base/compile'.  Stop. 
lede/include/toplevel.mk:228: recipe for target 'package/feeds/luci/luci-base/compile' failed