coolsnowwolf/lede #5367: dns2socks:编译出错

kongfl888, 2020-08-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：dns2socks编译出错，下载的包MD5不匹配，版本可能更新了

### (3) 详细日志

A：

https://downloads.sourceforge.net/dns2socks/SourceCode.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Dload  Upload   Total   Spent    Left  Speed
 
0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0

100   746  100   746    0     0   3674      0 --:--:-- --:--:-- --:--:--  3674
Hash of the downloaded file does not match (file:**afd33cb888461aae3bfd05b80c9e42a8**, requested: ec82de936ad004cc940502cd2a1bff5b) - deleting download.

Mattraks, 2020-08-27:
直接下载上传上去,重定向问题

kongfl888, 2020-08-27:
代码更新周期长，文件不足百K，是否考虑下项目内置，或者使用GitHub源
比如这个：https://github.com/kongfl888/dns2socks

closed 2020-08-31 by kongfl888