coolsnowwolf/lede #5371: target-x86_64_musl issue

vincentdxu, 2020-08-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

 2. 详细叙述

(1) 具体问题

A：make -j8 download V=s 后 总有几条出错 都与 target-x86_64_musl 相关，请教有何解决方法

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

make[2]: Leaving directory '/home/vincent/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/download#0.01#0.00#0.09
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/vincent/lede/toolchain/fortify-headers'
time: toolchain/fortify-headers/download#0.01#0.00#0.09
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/vincent/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/download#0.01#0.00#0.12
**bash: x86_64-openwrt-linux-musl-gcc: command not found
bash: x86_64-openwrt-linux-musl-gcc: command not found
bash: x86_64-openwrt-linux-musl-gcc: command not found**
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/vincent/lede/toolchain/musl'
time: toolchain/musl/download#0.03#0.01#0.14
make[1]: Leaving directory '/home/vincent/lede'
make[1]: Entering directory '/home/vincent/lede'

make[2]: Leaving directory '/home/vincent/lede/package/kernel/exfat'
time: package/kernel/exfat/download#0.14#0.01#1.19
make[2]: Entering directory '/home/vincent/lede/package/lean/adbyby'
**can't open file '/home/vincent/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.59/.config' at /home/vincent/lede/scripts/kconfig.pl line 32.**
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/vincent/lede/feeds/packages/net/nlbwmon'
time: package/feeds/packages/nlbwmon/download#0.33#0.07#3.73
make[2]: Entering directory '/home/vincent/lede/package/lean/autocore'

make[2]: Leaving directory '/home/vincent/lede/package/network/utils/ipset'
time: package/network/utils/ipset/download#0.13#0.03#1.24
**grep: /home/vincent/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.59/.config: No such file or directory**
make[2]: Entering directory '/home/vincent/lede/package/network/utils/resolveip'
make[2]: Entering directory '/home/vincent/lede/package/system/ca-certificates'
make[2]: Nothing to be done for 'download'.

BecodReyes, 2020-08-29:
疑似linux内核下载不完全
国内请尽量全局科学上网

AmadeusGhost, 2020-08-30:
make download 这边提示这个没关系的，大不了 make clean 即可

closed 2021-06-20 by coolsnowwolf