coolsnowwolf/lede #5373: 编译出来的ffmpeg因为内核版本过高无法安装

Gokou-Ruri, 2020-08-28:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译ffmpeg为ipk，无法安装。

### (2) 路由器型号和固件版本

Newifi-D2
OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13)
内核版本 5.4.59

### (3) 详细日志

A：
```
Installing ffmpeg (4.3-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ffmpeg:
 *      kernel (= 5.4.59-1-66fd27937b6084baec5a148626b74960)
 * opkg_install_cmd: Cannot install package ffmpeg.
```

RayorRui, 2020-08-28:
不是版本过高，是后面一串字母和内核对不上

AmadeusGhost, 2020-08-29:
这种编译进去即可，跟内核版本没关系

Gokou-Ruri, 2020-08-30:
> 这种编译进去即可，跟内核版本没关系

内置的Flash空间太小，所以想编译成ipk再安装。

AmadeusGhost, 2020-08-30:
> 内置的Flash空间太小，所以想编译成ipk再安装。

那就换个路由器或者 `opkg install xxx --force-depends` 或者 `--force-checksum`，这个确实不是 bug

Gokou-Ruri, 2020-08-31:
> > 内置的Flash空间太小，所以想编译成ipk再安装。
> 
> 那就换个路由器或者 `opkg install xxx --force-depends` 或者 `--force-checksum`，这个确实不是 bug

谢谢大佬指导！现在安装成功了。

closed 2020-08-31 by Gokou-Ruri