coolsnowwolf/lede #5375: 最新更新的OP无法使用docker

qq1316237299, 2020-08-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) [具体问题](url
[image]
[image]

)

A：最新更新的OP无法使用docker  5.4.59 

OpenWrt R20.8.27 / LuCI Master (git-20.191.36863-eee6bae)
--

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Sat Aug 29 03:12:26 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:28 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:30 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:32 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:34 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:36 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:38 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:40 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:42 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:44 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:46 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:48 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:50 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:52 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:53 2020 daemon.err dockerd[5870]: time="2020-08-29T03:12:53.177765233+08:00" level=warning msg="Your kernel does not support Block I/O weight or the cgroup is not mounted. Weight discarded."
Sat Aug 29 03:12:54 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.052836] docker0: port 1(vethb1cdd4f) entered blocking state
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.059194] docker0: port 1(vethb1cdd4f) entered disabled state
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.065833] device vethb1cdd4f entered promiscuous mode
Sat Aug 29 03:12:54 2020 daemon.err dockerd[5870]: time="2020-08-29T03:12:54.924243196+08:00" level=warning msg="Failed to allocate and map port 6888-6888:  (iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 6888 -j DNAT --to-destination 172.17.0.2:6888 ! -i docker0: iptables: No chain/target/match by that name.\n (exit status 1))"
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.097019] docker0: port 1(vethb1cdd4f) entered disabled state
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.106987] device vethb1cdd4f left promiscuous mode
Sat Aug 29 03:12:54 2020 kern.info kernel: [ 1545.112618] docker0: port 1(vethb1cdd4f) entered disabled state
Sat Aug 29 03:12:55 2020 daemon.err dockerd[5870]: time="2020-08-29T03:12:55.002213538+08:00" level=error msg="87a8cd03655fe5ea892f69ff45824a3abef1b55a0981bb289fde84b310f83602 cleanup: failed to delete container from containerd: no such container"
Sat Aug 29 03:12:55 2020 daemon.err dockerd[5870]: time="2020-08-29T03:12:55.002263948+08:00" level=error msg="Handler for POST /containers/aria2-pro/start returned error: driver failed programming external connectivity on endpoint aria2-pro (ab978ae016b6e1441c77f20b2224b81716db514cb6924b976f3e47f07582386a):  (iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 6888 -j DNAT --to-destination 172.17.0.2:6888 ! -i docker0: iptables: No chain/target/match by that name.\n (exit status 1))"
Sat Aug 29 03:12:56 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:12:58 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:00 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:02 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:04 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:06 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:08 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:10 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:12 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:14 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:16 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:18 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:20 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:22 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:24 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:26 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:28 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.862624] docker0: port 1(vethe18b879) entered blocking state
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.869320] docker0: port 1(vethe18b879) entered disabled state
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.876086] device vethe18b879 entered promiscuous mode
Sat Aug 29 03:13:30 2020 daemon.err dockerd[5870]: time="2020-08-29T03:13:30.731347510+08:00" level=warning msg="Failed to allocate and map port 6888-6888:  (iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 6888 -j DNAT --to-destination 172.17.0.2:6888 ! -i docker0: iptables: No chain/target/match by that name.\n (exit status 1))"
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.903027] docker0: port 1(vethe18b879) entered disabled state
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.914421] device vethe18b879 left promiscuous mode
Sat Aug 29 03:13:30 2020 kern.info kernel: [ 1580.920069] docker0: port 1(vethe18b879) entered disabled state
Sat Aug 29 03:13:30 2020 daemon.err dockerd[5870]: time="2020-08-29T03:13:30.807347519+08:00" level=error msg="87a8cd03655fe5ea892f69ff45824a3abef1b55a0981bb289fde84b310f83602 cleanup: failed to delete container from containerd: no such container"
Sat Aug 29 03:13:30 2020 daemon.err dockerd[5870]: time="2020-08-29T03:13:30.807394933+08:00" level=error msg="Handler for POST /containers/aria2-pro/start returned error: driver failed programming external connectivity on endpoint aria2-pro (67d98749ad88db56c171541df5ee5e6936035f4b01e247480f4aa8b4c7c3b606):  (iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 6888 -j DNAT --to-destination 172.17.0.2:6888 ! -i docker0: iptables: No chain/target/match by that name.\n (exit status 1))"
Sat Aug 29 03:13:30 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:32 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:34 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:36 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:38 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:40 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:42 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 03:13:44 2020 daemon.err netdata[6142]: PROCFILE: Cannot open file '/proc/sysvipc/shm'

lexin8, 2020-08-29:
docker run -d -p 9000:9000 -v /var/run/docker.sock:/var/run/docker.sock --restart=always --name prtainer portainer/portainer

qq1316237299, 2020-09-03:
> docker run -d -p 9000:9000 -v /var/run/docker.sock:/var/run/docker.sock --restart=always --name prtainer portainer/portainer

铁子这个也打不开的呢 
[image]

closed 2020-09-10 by qq1316237299