coolsnowwolf/lede #5384: netdata无限报错

Chuanwei2001, 2020-08-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
netdata无限报错
A：

### (2) 路由器型号和固件版本
OpenWrt R20.8.27 / LuCI Master 
A：

### (3) 详细日志
Sat Aug 29 23:24:09 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:11 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:13 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:15 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:17 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:19 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:21 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:23 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:25 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:27 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:29 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:31 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Sat Aug 29 23:24:33 2020 daemon.err netdata[3850]: PROCFILE: Cannot open file '/proc/sysvipc/shm
A：

hcym, 2020-08-29:
从来不编译这个玩意，x86上面都能看到cpu占用，mtk的估计容易卡死

mysteryman2045, 2020-08-29:
你确定你“是否搜索了issue (使用 "x" 选择)[x ] 没有类似的issue”？？？

cs-star, 2020-08-30:
删除netdata 相关软件包

FusionPlmH, 2020-08-31:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> netdata无限报错
> A：
>
> ### (2) 路由器型号和固件版本
> OpenWrt R20.8.27 / LuCI Master
> A：
>
> ### (3) 详细日志
> Sat Aug 29 [...]

我也有一模一樣的

closed 2021-06-20 by coolsnowwolf