coolsnowwolf/lede #5387: libustream-openssl和libustream-mbedtls编译失败

Dimitri2020007, 2020-08-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
#在线编译脚本不管选了libustream-openssl或者libustream-mbedtls或者都不选总会编译失败（编译时候我2选1）
#查看编译日志就是和这两个组件有关我也不知道是咋回事

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

2020-08-29T18:48:31.9713300Z make[3]: Entering directory '/workdir/openwrt/package/libs/ustream-ssl'
2020-08-29T18:48:32.4981015Z mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2020-08-29T18:48:32.5009352Z echo '20200215' | cmp -s - /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libustream-mbedtls.version || echo '20200215' > /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libustream-mbedtls.version
2020-08-29T18:48:32.5044865Z SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/ustream-ssl-mbedtls/ustream-ssl-2020-03-13-5e1bc342/.pkgdir/libustream-mbedtls/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2020-08-29T18:48:32.5091928Z touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libustream-mbedtls_installed
2020-08-29T18:48:32.5189298Z echo "libustream-mbedtls" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/ustream-ssl.mbedtls.install
2020-08-29T18:48:32.5207010Z make[3]: Leaving directory '/workdir/openwrt/package/libs/ustream-ssl'
2020-08-29T18:48:32.5216889Z time: package/libs/ustream-ssl/mbedtls/compile#0.42#0.14#0.55
2020-08-29T18:48:32.5349234Z make[3]: Entering directory '/workdir/openwrt/package/libs/ustream-ssl'
2020-08-29T18:48:33.0580176Z mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2020-08-29T18:48:33.0609583Z echo '20200215' | cmp -s - /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libustream-openssl.version || echo '20200215' > /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libustream-openssl.version
2020-08-29T18:48:33.0642628Z SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/ustream-ssl-openssl/ustream-ssl-2020-03-13-5e1bc342/.pkgdir/libustream-openssl/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2020-08-29T18:48:33.0701358Z touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libustream-openssl_installed
2020-08-29T18:48:33.0804358Z echo "libustream-openssl" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/ustream-ssl.openssl.install
2020-08-29T18:48:33.0826625Z make[3]: Leaving directory '/workdir/openwrt/package/libs/ustream-ssl'

2020-08-29T19:28:11.1538555Z Collected errors:
2020-08-29T19:28:11.1539936Z Configuring iwlwifi-firmware-iwl8265.
2020-08-29T19:28:11.1541415Z  * check_data_file_clashes: Package libustream-openssl20200215 wants to install file /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
2020-08-29T19:28:11.1542618Z Configuring luci-i18n-verysync-zh-cn.
2020-08-29T19:28:11.1543062Z 	But that file is already provided by package  * libustream-mbedtls20200215
2020-08-29T19:28:11.1543350Z  * opkg_install_cmd: Cannot install package libustream-openssl20200215.
2020-08-29T19:28:11.1543847Z  * resolve_conffiles: Existing conffile /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
2020-08-29T19:28:11.1543999Z make[2]: *** [package/install] Error 255

AmadeusGhost, 2020-08-30:
你自己软件包选冲突了吧

Dimitri2020007, 2020-08-30:
> 你自己软件包选冲突了吧

我在线编译脚本只选了libustream-mbedtls，脚本自己处理的时候就把2个一起选上了，我也不知道这是咋回事，相关的2个软件包我已经取消了。

AmadeusGhost, 2020-08-30:
默认是用的 libustream-openssl 吧

Dimitri2020007, 2020-08-30:
#一开始我的编译脚本只选了libustream-openssl，后来说软件包冲突，libustream-openssl已经包含在libustream-mbedtls中。
#后来我脚本中只选了libustream-mbedtls，还是会出错。脚本处理的时候依旧把2个一起选上了。我就不知道咋处理了。
#luci-nginx-ssl都需要哪些依赖软件啊？和openwrt官网上说的一样吗？还是要选别的？

zheshifandian, 2020-08-30:
make dirclean && make -j8 V=s

Dimitri2020007, 2020-08-31:
> make dirclean && make -j8 V=s

不是那个问题

Dimitri2020007, 2020-08-31:
luci-nginx-ssl换别的ssl一样还是报错

2020-08-30T22:26:49.9122064Z Collected errors:
2020-08-30T22:26:49.9122743Z Configuring kmod-ipt-ipmark.
2020-08-30T22:26:49.9124156Z  * check_data_file_clashes: Package libustream-openssl20200215 wants to install file /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
2020-08-30T22:26:49.9124435Z Configuring ath10k-firmware-qca988x.
2020-08-30T22:26:49.9125863Z 	But that file is already provided by package  * libustream-mbedtls20200215
2020-08-30T22:26:49.9126105Z Configuring tor-geoip.
2020-08-30T22:26:49.9127123Z  * opkg_install_cmd: Cannot install package libustream-openssl20200215.
2020-08-30T22:26:49.9127359Z Configuring coreutils-sha384sum.
2020-08-30T22:26:49.9128558Z  * resolve_conffiles: Existing conffile /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
2020-08-30T22:26:49.9128820Z Configuring kmod-be2net.
2020-08-30T22:26:49.9129734Z make[2]: *** [package/install] Error 255

Dimitri2020007, 2020-09-01:
是加密工具的问题，谢谢大家了

closed 2020-09-01 by Dimitri2020007