coolsnowwolf/lede #5390: X64软路由按电源键**无法关机**（内核版本5.4）

Nicepaul, 2020-08-31:
X64 3125U 工控机软路由
内核版本5.4下，  按电源开关键 无法关机（不正常）
内核版本4.19下，按电源开关键 可以关机（正常）
系统中没有日志
请L大看看是否可以修正

hcym, 2020-08-31:
拔线啊，那有耐心按键，我的是5020u的pc，还有蜗牛章鱼和n1，一堆斐讯们都是拔下来插进去😥

FusionPlmH, 2020-08-31:
> X64 3125U 工控机软路由
> 内核版本5.4下， 按电源开关键 无法关机（不正常），只能长按电源键强制关机
> 内核版本4.19下，按电源开关键 可以关机（正常）
> 系统中没有日志
> 请L大看看是否可以修正

連接hdmi看按關機的反應

Nicepaul, 2020-08-31:
> 關機

刚试了，一样的，电源键无法关机。

hxlls, 2020-08-31:
路由没必要关机把,路由本身设计就是24小时运行的,好像所有厂商的设计都是,连关机按键都没有

martin0607, 2020-09-01:
ssh --> poweroff

Nicepaul, 2020-09-01:
> ssh --> poweroff

ssh关机是可以的。但是路由器是给别人用的，他有每天关机的习惯，而且也不会ssh，所有希望能用电源键关机。老是强制关机总感觉不是太好

martin0607, 2020-09-01:
> > ssh --> poweroff
> 
> ssh关机是可以的。但是路由器是给别人用的，他有每天关机的习惯，而且也不会ssh，所有希望能用电源键关机。老是强制关机总感觉不是太好

可以搞个cron脚本，定时关机

ihuaer, 2020-09-01:
> > ssh --> poweroff
> 
> ssh关机是可以的。但是路由器是给别人用的，他有每天关机的习惯，而且也不会ssh，所有希望能用电源键关机。老是强制关机总感觉不是太好

`菜单里添加关机
[image]
`

Nicepaul, 2020-09-02:
解决了，安装acpid就可以了。

closed 2020-09-02 by Nicepaul