coolsnowwolf/lede #5391: 游戏模式UDP中继服务器游戏内语音无法使用

lsbjkc, 2020-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

开启“游戏模式UDP中继服务器”，pubg无法正常语音，听不到队友的麦克风语音，换加速器后正常

### (2) 路由器型号和固件版本

固件版本 OpenWrt R20.7.20 / LuCI Master (git-20.223.26773-d18ef13) 
路由器型号 Intel(R) Celeron(R) CPU J1900

### (3) 详细日志

Mon Aug 31 12:55:13 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:15 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:17 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:19 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:21 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:23 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:25 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:27 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:29 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:31 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:33 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:35 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:37 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:39 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:41 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:43 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:45 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:47 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:49 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:51 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:53 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 12:55:55 2020 daemon.err netdata[5044]: PROCFILE: Cannot open file '/proc/sysvipc/shm'
Mon Aug 31 13:08:14 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Aug 31 13:08:14 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Aug 31 13:08:17 2020 daemon.info dnsmasq[11597]: Connected to system UBus
Mon Aug 31 13:08:17 2020 cron.err crond[11632]: crond (busybox 1.31.1) started, log level 5
Mon Aug 31 13:08:17 2020 kern.info /usr/bin/ssr-redir[11877]: initializing ciphers... none
Mon Aug 31 13:08:17 2020 kern.info /usr/bin/ssr-redir[11879]: initializing ciphers... none
Mon Aug 31 13:08:17 2020 kern.info /usr/bin/ssr-redir[11881]: initializing ciphers... none
Mon Aug 31 13:08:17 2020 kern.info /usr/bin/ssr-redir[11883]: initializing ciphers... none
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11877]: tcp port reuse enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11877]: listening at 0.0.0.0:1234
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11877]: running from root user
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11879]: tcp port reuse enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11879]: listening at 0.0.0.0:1234
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11879]: running from root user
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11881]: tcp port reuse enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11881]: listening at 0.0.0.0:1234
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11881]: running from root user
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11883]: tcp port reuse enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11883]: listening at 0.0.0.0:1234
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11883]: running from root user
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: initializing ciphers... none
Mon Aug 31 13:08:18 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Aug 31 13:08:18 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: UDP relay enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: udp port reuse enabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: TCP relay disabled
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: listening at 0.0.0.0:1234
Mon Aug 31 13:08:18 2020 kern.info /usr/bin/ssr-redir[11923]: running from root user
Mon Aug 31 13:08:21 2020 daemon.info dnsmasq[12077]: Connected to system UBus

BGzErol, 2020-09-05:
我观察了下确实有这个问题，不知道这些连不上语音的游戏关于队内语音的部分是不是用了quic协议，ssrp是默认屏蔽quic的，如果是的话那用加速器或netch吧，毕竟屏蔽quic是刚需

closed 2021-06-20 by coolsnowwolf

etsam, 2022-08-05:
跟你一样的问题 解决了吗

lsbjkc, 2022-08-05:
> 跟你一样的问题 解决了吗

没有啊，用SSTAP这种虚拟网卡类的代理软件才能语音