coolsnowwolf/lede #5392: SSR+ 绕开大陆IP模式 访问控制(黑/白名单) 不稳定

kkfuzi, 2020-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：软路由刷openwrt
SSR+ 模式：绕开大陆IP
访问控制(黑/白名单) 不稳定，已经设置成白名单的域名仍然走代理。
尝试换成GWF模式，切回来正常，修改域名或者换节点重新保存又不工作。

### (2) 路由器型号和固件版本

A：OpenWrt R20.8.27

fhbyljj, 2020-09-12:
我没有这个问题，过一会就生效了

kkfuzi, 2020-09-12:
> 我没有这个问题，过一会就生效了

找到问题原因了。
SSR+搭配AdGuard Home，SSR+模式 绕过大陆IP，DNS解析是PDNSD
如果ADH设置成“使用53端口替换 dnsmasq” 或者“重定向53端口到AdGuardHome”模式， SSR+的访问控制/白名单 无效，设置成白名单的国外域名依然走代理。
如果ADH设置成“作为dnsmasq的上游服务器” 就没问题了。。。

但用上游服务器模式就没法统计局域网客户端信息和自定义客户端的广告过滤。。

确定不是openwrt和ssr+的问题了，考虑一下怎么从AGH解决这个问题。。。

kkfuzi, 2020-09-18:
得到信息了
https://github.com/AdguardTeam/AdGuardHome/issues/1191
AGH官方的v0.104.0-beta1开发进程里边：“7931e50 + DNS: add "ipset" configuration setting”就是在做这个优化，只能坐等了。

closed 2020-09-18 by kkfuzi