coolsnowwolf/lede #5394: 新版本wifi 设置问题

Guxing0321, 2020-08-31:
Hi team，

我发现新版本的wifi 设置变化很大
[image]

[image]

我现在有点困惑，不知道哪个需要改动

1. 设置地区的那种，我发现没有china，
2. 设置wifi 信号道，不知道该怎么选，之前是直接选择china 就出来了
3. 设置wifi 功率问题， 高级设置里面有个 TX Power ，默认是100 ，基本设置里面还需要改吗
4.  网络 - 无线  有个接口如下，这是什么，可以去掉吗
         接口: apclii0 | 类型: STA | 状态: Disconnected 
         Wireless is disabled or not associated

BR
Robin

asushugo, 2020-09-01:
MTK闭源驱动的是这样的

Guxing0321, 2020-09-01:
hi asushugo,

我用的是 newifi3 D2 的硬件
之前也编译过很多次，wifi 设置不是这个样子的
是刚改了，还是我编译步骤有问题？

BR
Robin

EnnawYang, 2020-09-01:
https://github.com/coolsnowwolf/lede/issues/5002

Guxing0321, 2020-09-01:
Hi EnnawYang,

好的，收到，我在那个单子里面回复

BR
Robin

closed 2020-09-03 by Guxing0321